google-deepmind/mujoco #2416: Specify a class in MjSpec when adding elements

vittorione94, 2025-02-09:
### The feature, motivation and pitch

Hello,

When procedurally generating a file using MjSpec currently there is no option to specify a dclass (something that was possible to do in mjcf). 

For example if I'm loading an xml file that has some predefined classes like <my_muscle> I would like to reuse that class in when using:
`spec.add_actuator(...., dclass='my_muscle')`

As it is right now I would need to pass all the attributes of the class to add_actuator and specifying a class loses its significance. 

Would it be possible to add this functionality to all the <add_> functions (this can be extended to geoms, bodies, ect.)

Thank you in advance!

quagla, 2025-02-10:
Hi, defaults can already be specified as an argument, see https://github.com/google-deepmind/mujoco/blob/main/python/mujoco/specs_test.py#L575 is this what you wanted?

yuvaltassa, 2025-02-10:
Maybe we should rename the argument?

quagla, 2025-02-12:
1. Use default as keyword argument
2. You need motor_def.actuator
3. The name of the main default cannot be changed
4. Only bodies have children, that's why only bodies have childclass, see https://mujoco.readthedocs.io/en/latest/XMLreference.html#body-childclass

quagla, 2025-02-13:
Closing for now but let me know if you have other questions.