google-deepmind/mujoco #2420: MJX collision between Ellipsoid / Cylinder and Hfield

youwyu, 2025-02-11:
### The feature, motivation and pitch

As shown in [MJX 3.1.6](https://mujoco.readthedocs.io/en/3.1.6/mjx.html), ELLIPSOID and CYLINDER are implemented but only collide with other primitives.

hello-robot-shehab, 2025-03-21:
Hi @yuvaltassa, I didn't want to open a duplicate/related issue; do you know if supporting cylinder to mesh geom collisions in MJX is planned to be supported?

We're trying to use this mobile manipulator from the Mujoco Menagerie https://github.com/google-deepmind/mujoco_menagerie/tree/main/hello_robot_stretch_3 with MJX, but the main issue with using this MJCF are the cylinders used for wheel collision. Like @youwyu says, using a mesh cylinders for wheels is no-go.

Thank you for your thoughts on what we could do here!

Side-question: I don't know much about how these geom colliders are built, but I see sphere <> mesh collisions implemented, is there a way to re-purpose that sphere implementation for cylinders? Thanks!

scott-yj-yang, 2025-05-06:
Hi MuJoCo team, Is there an estimated timeline for enabling full collision support between primitive geoms—especially ellipsoids—and heightfields in MJX? We would greatly appreciate any updates on this feature, as it will help us better plan training for our biomechanical model, which relies heavily on ellipsoids for body construction.

hello-robot-shehab, 2025-05-06:
We're also still waiting on this, would love to know if this will be supported soon. Thanks Mujoco team!

thowell, 2025-09-15:
- ellipsoid<>cylinder
- cylinder<>mesh
- ellipsoid<>hfield

are available in MJX via the MuJoCo Warp backend (set `impl='warp'`)

```
mx = mjx.put_model(m, impl='warp')
dx = mjx.put_data(m, impl='warp')
```

Abban-Fahim, 2025-12-30:
@thowell The warp backend indeed does support these collisions, but when I try to use it in Playground with hfields or even just a few cubes, the initial compilation always crashes, most likely due to the limited amount of RAM (not VRAM) I have (16 GB). As such, I can only use MJX with my current hardware and deadline, are there still any plans for supporting these collisions? Alternatively, do you have any pointers on how I could implement this myself (I'd be happy to contribute), or if a rudimentary implementation already?

btaba, 2026-01-02:
This may be a fix for hfield https://github.com/google-deepmind/mujoco_warp/commit/55ab24595ea4939ff583f1910e1615361d49d9ad worth trying out again. If there are memory issues with the Warp backend, feel free to report in the mujoco_warp repo. There are no plans to support these collisions in the JAX implementaiton.