google-deepmind/mujoco #2421: mjSpec find_geom not implemented

vittorione94, 2025-02-12:
### The feature, motivation and pitch

Hello,

when I try to search for a specific geom using mjSpec like:
`spec.find_geom("my_geom")`

I get:
`AttributeError: 'mujoco._specs.MjSpec' object has no attribute 'find_geom'`

Would it be possible to add it?

kevinzakka, 2025-02-12:
Commit cef5b31 dropped the `find` prefix from these methods, you'll be able to do the same thing we do with MjModels soon. Will be out when the next release is cut.

closed 2025-02-12 by yuvaltassa

vittorione94, 2025-02-12:
@kevinzakka I'm using 3.2.7 so add_geom has been added recently?

yuvaltassa, 2025-02-12:
Yes in e54f3c222de91e6ef8d553395e8ee8604f4ce946

Subsequently the `find_` was dropped in cef5b31a3fb83490ab70d1da87880633f1507a87