google-deepmind/mujoco #2444: `contact/exclude` appears to ignore links which are rigidly attached to the worldbody

rdeits-bd, 2025-02-21:
### Intro

I'm a researcher at Boston Dynamics. 

### My setup

I'm using the latest release 3.2.7 for linux x86_64. 

### What's happening? What did you expect?

Here's a minimal model that illustrates the problem. First, a working version:

```xml
<mujoco model="minimal-repro">
  <worldbody>
    <body name="body_1">
      <freejoint/>
      <geom type="sphere" size="0.5" mass="1"/>
      <body name="body_2" pos="0.25 0 0">
        <joint type="slide" axis="1 0 0"/>
        <geom type="sphere" size="0.5" mass="0.5"/>
        <body name="body_3" pos="0.25 0 0">
          <joint type="slide" axis="1 0 0"/>
          <geom type="sphere" size="0.5" mass="0.5"/>
        </body>
      </body>
    </body>
  </worldbody>
  <contact>
    <exclude body1="body_1" body2="body_3"/>
  </contact>
</mujoco>
```

This model consists of three spheres, each of which overlaps with the others. Collision between body_1 and body_2 is ignored because they are direct ancestors in the kinematic tree, likewise body_2 and body_3. I have manually excluded collision between body_1 and body_3 with a `contact/exclude` group. This all works correctly: When I run `./bin/simulate minimal-repro.xml`, the three spheres fall together. 

However, if I remove the `freejoint`, like so:

```xml
<mujoco model="minimal-repro">
  <worldbody>
    <body name="body_1">
      <geom type="sphere" size="0.5" mass="1"/>
      <body name="body_2" pos="0.25 0 0">
        <joint type="slide" axis="1 0 0"/>
        <geom type="sphere" size="0.5" mass="0.5"/>
        <body name="body_3" pos="0.25 0 0">
          <joint type="slide" axis="1 0 0"/>
          <geom type="sphere" size="0.5" mass="0.5"/>
        </body>
      </body>
    </body>
  </worldbody>
  <contact>
    <exclude body1="body_1" body2="body_3"/>
  </contact>
</mujoco>
```

then body_3 immediately shoots away, presumably due to the collision with body_1. 

I have also tried `<compiler fusestatic="false"/>` (even though I expect that to be the default), and it had no effect. 

### Steps for reproduction

1. Load the model below with the `./bin/simulate` from the latest Mujoco release (3.2.7)
2. Observe that the sphere flies away.

### Minimal model for reproduction

```xml
<mujoco model="minimal-repro">
  <worldbody>
    <body name="body_1">
      <geom type="sphere" size="0.5" mass="1"/>
      <body name="body_2" pos="0.25 0 0">
        <joint type="slide" axis="1 0 0"/>
        <geom type="sphere" size="0.5" mass="0.5"/>
        <body name="body_3" pos="0.25 0 0">
          <joint type="slide" axis="1 0 0"/>
          <geom type="sphere" size="0.5" mass="0.5"/>
        </body>
      </body>
    </body>
  </worldbody>
  <contact>
    <exclude body1="body_1" body2="body_3"/>
  </contact>
</mujoco>
```

### Confirmations

- [x] I searched the [latest documentation](https://mujoco.readthedocs.io/en/latest/overview.html) thoroughly before posting.
- [x] I searched previous [Issues](https://github.com/google-deepmind/mujoco/issues) and [Discussions](https://github.com/google-deepmind/mujoco/discussions), I am certain this has not been raised before.

yuvaltassa, 2025-02-21:
Thanks!

Can you please try to use "world" as the excluded body and report what happens?

IIRC, bodies that are kinematically welded to the world are considered to be the world for collision purposes. If my suggestion above works then this is basically a documentation issue. If not, we will examine more carefully.

yuvaltassa, 2025-02-21:
Mmmm this is [technically documented](https://mujoco.readthedocs.io/en/latest/computation/index.html#selection) but is rather hard to find. Assuming my diagnosis is correct, the `exclude` documetation should be clarified to explain that in cases of static descendants, it is the top ancestor body that should be excluded.

rdeits-bd, 2025-02-21:
Interestingly, setting `world` instead of `body_1` actually results in yet another behavior -- now both the second and third spheres fly away:

```
<mujoco model="minimal-repro">
  <compiler fusestatic="false"/>
  <worldbody>
    <body name="body_1">
      <geom type="sphere" size="0.5" mass="1"/>
      <body name="body_2" pos="0.25 0 0">
        <joint type="slide" axis="1 0 0"/>
        <geom type="sphere" size="0.5" mass="0.5"/>
        <body name="body_3" pos="0.25 0 0">
          <joint type="slide" axis="1 0 0"/>
          <geom type="sphere" size="0.5" mass="0.5"/>
        </body>
      </body>
    </body>
  </worldbody>
  <contact>
    <exclude body1="world" body2="body_3"/>
  </contact>
</mujoco>
```

[image]

vs what I got with `body1="body_1"`:

[image]

yuvaltassa, 2025-02-22:
Thanks, I'll investigate.

yuvaltassa, 2025-03-08:
Okay I finally took a look.

1. My recommendation was wrong, you can and should use the names of static bodies in exclude.
2. Your diagnosis of what happens in your second example 

> Observe that the sphere flies away.

is incorrect, it is not body2 that flies away, but body3. It helps if you turn on body names in the visualizer.

[image]

So as far as I can tell this is a user error and everything works as intended and documented.

Please let me know if I am mistaken. Closing for now.

rdeits-bd, 2025-03-13:
You're right, my mistake. In fact, this is almost word-for-word the "Surprising Collisions" section from the docs, which I hadn't found before now: https://mujoco.readthedocs.io/en/stable/overview.html#surprising-collisions

closed 2025-07-21 by yuvaltassa