google-deepmind/mujoco #2463: Confusing flattening of bind shapes

hartikainen, 2025-02-26:
### Intro

Hi!

### My setup

$ python -c "import mujoco; print(mujoco.__version__)"
3.2.7

### What's happening? What did you expect?

When binding elements with array of length 1, the leading shape gets flattened. I find this extremely confusing and would expect the size of the leading dimension to match the size of the binded array. I took a cursory look on the documentation and didn't see a mention of such behavior so I suppose this might be a bug.

See example below.

### Steps for reproduction

```xml
from absl import app
import mujoco
from mujoco import mjx
import numpy as np

def main(args):
    xml=r"""
<mujoco>
  <worldbody>
    <body name="main">
      <body name="box">
        <joint name="box" type="hinge" range="-1 +1"/>
        <geom name="box" size="0.15 0.15 0.15" mass="1" type="box"/>
      </body>
      <body name="sphere">
        <joint name="sphere" type="hinge" range="-1 +1"/>
        <geom name="sphere" size="0.15 0.15 0.15" mass="1" type="box"/>
      </body>
    </body>
  </worldbody>
</mujoco>
"""

    spec = mujoco.MjSpec.from_string(xml)

    joint_box = next(j for j in spec.joints if j.name == "box")
    joint_sphere = next(j for j in spec.joints if j.name == "sphere")
    joints = [joint_box, joint_sphere]

    mj_model = spec.compile()
    mj_data = mujoco.MjData(mj_model)
    mjx_model = mjx.put_model(mj_model)
    mjx_data = mjx.put_data(mj_model, mj_data)

    joint_ids = mjx_data.bind(mjx_model, joints).id
    print(f"{joint_ids=}; {type(joint_ids)=}")
    assert np.shape(joint_ids) == (2,), np.shape(joint_ids)

    joint_ids = mjx_data.bind(mjx_model, joints[0:1]).id
    print(f"{joint_ids=}; {type(joint_ids)=}")
    assert np.shape(joint_ids) == (1,), np.shape(joint_ids)

if __name__ == "__main__":
    app.run(main)
```

```bash
$ python ./mujoco_flatten_bind.py
joint_ids=[0, 1]; type(joint_ids)=<class 'list'>
joint_ids=0; type(joint_ids)=<class 'int'>
Traceback (most recent call last):
  File "/Users/hartikainen/tmp/scripts/mujoco_flatten_bind.py", line 46, in <module>
    app.run(main)
  File "/Users/hartikainen/github/google-deepmind/mujoco/.venv/lib/python3.12/site-packages/absl/app.py", line 308, in run
    _run_main(main, args)
  File "/Users/hartikainen/github/google-deepmind/mujoco/.venv/lib/python3.12/site-packages/absl/app.py", line 254, in _run_main
    sys.exit(main(argv))
             ^^^^^^^^^^
  File "/Users/hartikainen/tmp/scripts/mujoco_flatten_bind.py", line 42, in main
    assert np.shape(joint_ids) == (1,), np.shape(joint_ids)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^
AssertionError: ()
```

### Confirmations

- [x] I searched the [latest documentation](https://mujoco.readthedocs.io/en/latest/overview.html) thoroughly before posting.
- [x] I searched previous [Issues](https://github.com/google-deepmind/mujoco/issues) and [Discussions](https://github.com/google-deepmind/mujoco/discussions), I am certain this has not been raised before.

hartikainen, 2025-02-26:
Looking at the support implementation, this seems like an explicit choice:
https://github.com/google-deepmind/mujoco/blob/929e94cf59ed5aec448ba549419cc1ccb41d23ea/mjx/mujoco/mjx/_src/support.py#L365-L366
https://github.com/google-deepmind/mujoco/blob/929e94cf59ed5aec448ba549419cc1ccb41d23ea/mjx/mujoco/mjx/_src/support.py#L441-L442

If so, this should be a feature request and not a bug report.

quagla, 2025-02-27:
Thanks for a nice MRE as usual! Would you like to create a PR directly?

quagla, 2025-02-27:
This was fixed in 1532cffdf0e04a5a734de8c0f16424acf84b2604

closed 2025-02-27 by copybara-service[bot]

hartikainen, 2025-02-27:
Hey. Apologies for bugging about this; I just tested this from `3.3.0` and it still seems to be failing. Not just with `id` but also with e.g. `qpos` and `qvel`. (I updated the original code snippet with `qpos` and `qvel`.)

yuvaltassa, 2025-03-30:
@hartikainen is this solved now?

hartikainen, 2025-03-30:
No, not from `mujoco==3.3.0` at least.

yuvaltassa, 2025-04-01:
Well sure, 3.3.0 was released on Feb 26, but @quagla 's  fix is from Feb 27.
Can you build from source and check again?
Or wait for the next release, should be out soonish

commit 2abb2e2d (2025-02-27): Add further MJX bind tests to clarify dimensions.

Fixes #2463.

PiperOrigin-RevId: 731674489
Change-Id: I9e21a2dd0b0636c94b2b917615889b9d721d542d