google-deepmind/mujoco #2464: Textures are not rendered on flexcomp mesh and gmsh types

jsll, 2025-02-27:
### Intro

Hi!

I am a researcher at KTH. I use MuJoCo for my research on deformable object manipulation.

### My setup

MuJoCo version 3.2.8

### What's happening? What did you expect?

When I apply texture to the flexcomp type mesh and gmsh the texture is not rendered.

Below is a screenshot of three flexcomp types: box, mesh, and gmesh, all with the same applied texture. Surprisingly, only the box model gets the texture applied. 

This issue is related to https://github.com/google-deepmind/mujoco/discussions/2013#discussioncomment-12225623

[image]

### Steps for reproduction

1. Unzip the attached zip folder
2. Load the model below.
3. run ./simulate path_to_model_below

### Minimal model for reproduction

Here are the assets folder that the model below reference to [assets.zip](https://github.com/user-attachments/files/19009723/assets.zip)

```
<mujoco model="texture_issue">

  <compiler meshdir="asset" texturedir="asset" />
  <option solver="CG" tolerance="1e-6" />

  <size memory="30M" />
  <option noslip_iterations="10">
    <flag multiccd="enable" />
  </option>

  <visual>
    <rgba haze="0.15 0.25 0.35 1"/>
    <quality shadowsize="4096"/>
    <map stiffness="700" shadowscale="0.5" fogstart="1" fogend="15" zfar="40" haze="1"/>
  </visual>

  <asset>
    <texture name="texsponge" type="2d" file="sponge.png" />
    <material name="matsponge" texture="texsponge" specular="0.3" />
    <texture name="texplane" type="2d" builtin="checker" rgb1=".2 .3 .4" rgb2=".1 0.15 0.2"
      width="512" height="512" mark="cross" markrgb=".8 .8 .8" />
    <material name="matplane" reflectance="0.3" texture="texplane" texrepeat="10 10"
      texuniform="true" />
  </asset>

  <worldbody>
    <light diffuse=".4 .4 .4" specular="0.1 0.1 0.1" pos="0 0 2.0" dir="0 0 -1" castshadow="false"/>
    <light directional="true" diffuse=".8 .8 .8" specular="0.2 0.2 0.2" pos="0 0 4" dir="0 0 -1"/>
    <body pos="0 0 1" name="body">
      <freejoint />
      <geom size=".1" contype="0" conaffinity="0" group="4" />
      <flexcomp name="softbox" type="box" count="7 7 7" spacing=".04 .04 .04"
        radius="0.01" dim="3" material="matsponge" dof="radial">
        <contact internal="false" selfcollide="none" />
        <edge equality="true" />
      </flexcomp>
    </body>
    <body
      pos="0.4 0 1" name="mesh">
      <freejoint />
      <geom size=".1" contype="0" conaffinity="0" group="4" />
      <flexcomp name="mesh" type="mesh" file="cloth.stl"
        radius="0.01" dim="2" material="matsponge">
        <contact internal="false" selfcollide="none" />
        <edge equality="true" />
      </flexcomp>
    </body>
    <body pos="0.6 0 1" name="gmsh">
      <freejoint />
      <geom size=".1" contype="0" conaffinity="0" group="4" />
      <flexcomp name="gmsh" type="gmsh" file="cloth.msh"
        radius="0.01" dim="2" material="matsponge" scale="0.1 0.1 0.1">
        <contact internal="false" selfcollide="none" />
        <edge equality="true" />
      </flexcomp>
    </body>
    <geom name="ground" type="plane" size="0 0 1" pos="0 0 0" quat="1 0 0 0" material="matplane"
      condim="1" />
  </worldbody>
</mujoco>
```

### Code required for reproduction

### Confirmations

- [x] I searched the [latest documentation](https://mujoco.readthedocs.io/en/latest/overview.html) thoroughly before posting.
- [x] I searched previous [Issues](https://github.com/google-deepmind/mujoco/issues) and [Discussions](https://github.com/google-deepmind/mujoco/discussions), I am certain this has not been raised before.

jsll, 2025-02-27:
Based on the fix for rendering texture on flexcomp grid types from here https://github.com/google-deepmind/mujoco/commit/6a862478103ee7c618b2a1f9b84c6f21fcc5d882, I think these two methods need to be modified https://github.com/google-deepmind/mujoco/blob/2abb2e2d5df35599d2df86188131f27f88996308/src/user/user_flexcomp.cc#L982  https://github.com/google-deepmind/mujoco/blob/2abb2e2d5df35599d2df86188131f27f88996308/src/user/user_flexcomp.cc#L1121

quagla, 2025-02-27:
Hi, for a box it's straightforward to compute the texture coordinates explicitly but for general meshes you have to provide the coordinates in the mesh file, see the lines that start `vt` in https://en.wikipedia.org/wiki/Wavefront_.obj_file

jsll, 2025-02-27:
[assets.zip](https://github.com/user-attachments/files/19012566/assets.zip)

Hello. Thanks for the input. So I change the mesh to an obj file and then I unwrapped. I also change the model to include a flexcomp box, flexcomp mesh, and geom mesh. Again, all have the same texture applied. As you can see in the image below, the texture now gets correctly applied to the flexcomp box and the geom mesh, but still not for the flexcomp mesh. Is there anything still that I have to do?

```
<mujoco model="texture_issue">

  <option solver="CG" tolerance="1e-6" />

  <size memory="30M" />
  <option noslip_iterations="10">
    <flag multiccd="enable" />
  </option>

  <visual>
    <rgba haze="0.15 0.25 0.35 1"/>
    <quality shadowsize="4096"/>
    <map stiffness="700" shadowscale="0.5" fogstart="1" fogend="15" zfar="40" haze="1"/>
  </visual>

  <asset>
    <texture name="texsponge" type="2d" file="sponge.png" />
    <material name="matsponge" texture="texsponge" specular="0.3" />
    <texture name="texplane" type="2d" builtin="checker" rgb1=".2 .3 .4" rgb2=".1 0.15 0.2"
      width="512" height="512" mark="cross" markrgb=".8 .8 .8" />
    <material name="matplane" reflectance="0.3" texture="texplane" texrepeat="10 10"
      texuniform="true" />
    <mesh name="test" file="cloth_extruded.obj" scale="1 1 1"/>
  </asset>

  <worldbody>
    <light diffuse=".4 .4 .4" specular="0.1 0.1 0.1" pos="0 0 2.0" dir="0 0 -1" castshadow="false"/>
    <light directional="true" diffuse=".8 .8 .8" specular="0.2 0.2 0.2" pos="0 0 4" dir="0 0 -1"/>
    <body pos="0 0 1" name="body">
      <freejoint />
      <geom size=".1" contype="0" conaffinity="0" group="4" />
      <flexcomp name="softbox" type="box" count="7 7 7" spacing=".04 .04 .04"
        radius="0.01" dim="3" material="matsponge" dof="radial">
        <contact internal="false" selfcollide="none" />
        <edge equality="true" />
      </flexcomp>
    </body>
    <body
      pos="0.4 0 1" name="objmesh">
      <freejoint />
      <geom size=".1" contype="0" conaffinity="0" group="4" />
      <flexcomp name="meshob" type="mesh" file="cloth_extruded.obj"
        radius="0.01" dim="2" material="matsponge">
        <contact internal="false" selfcollide="none" />
        <edge equality="true" />
      </flexcomp>
    </body>
    <body name="test">
      <freejoint/>
      <geom type="mesh" mesh="test" material="matsponge" pos="0.8 0 1" group="1" contype="0" conaffinity="0"/>
    </body>
    <geom name="ground" type="plane" size="0 0 1" pos="0 0 0" quat="1 0 0 0" material="matplane"
      condim="1" />
  </worldbody>
</mujoco>
```

[image]

commit 7651cb1b (2025-02-28): Copy texture coordinates from OBJ into flex. Fixes #2464.

PiperOrigin-RevId: 732109960
Change-Id: I506b3540ab5d489ce8f10d0c473919611ccb654a
closed 2025-02-28 by copybara-service[bot]