google-deepmind/mujoco #2477: Mesh Penetration Issue in MJX

Stillwtm, 2025-03-07:
### Intro

Hi!

I am a undergrad student at SJTU, I use MuJoCo for my research on robot manipulation.

### My setup

mujoco 3.3.0, mujoco-mjx 3.2.7, Python, Ubuntu 20.04

### What's happening? What did you expect?

I used the following command to load my model in MJX and visualize:
```
JAX_PLATFORM_NAME=cpu python -m mujoco.mjx.viewer --mjcf scene.xml
```
I noticed that a significant penetration occurs between the mesh of a bottle and a box.

https://github.com/user-attachments/assets/3be3db77-5a1c-4fdc-881f-1b1e090c0a13

However, when I removed three empty bodies from the model, this penetration issue seemed to disappear.

I am unsure why this behavior occurs, as my previous understanding was that empty bodies should not affect the physics simulation.

### Steps for reproduction

1. Run
```
JAX_PLATFORM_NAME=cpu python -m mujoco.mjx.viewer --mjcf scene.xml
```
2. Observe penetration between bottle mesh and the box
3. Comment out L72-74 in r0_real2sim.xml (as shown in r0_real2sim_1.xml).
```xml
      <!-- <body name='a' pos="1.05 0.25 0.97"/>
      <body name='b' pos="1.05 0.283 0.97"/>
      <body name='c' pos="1.05 0.217 0.97"/> -->
```
4. Run the viewer again; observe that the mesh penetration issue disappears.

### Minimal model for reproduction

[demo.zip](https://github.com/user-attachments/files/19123920/demo.zip)

### Confirmations

- [x] I searched the [latest documentation](https://mujoco.readthedocs.io/en/latest/overview.html) thoroughly before posting.
- [x] I searched previous [Issues](https://github.com/google-deepmind/mujoco/issues) and [Discussions](https://github.com/google-deepmind/mujoco/discussions), I am certain this has not been raised before.

gemcollector, 2025-03-11:
met the same issue.

btaba, 2025-03-31:
Hi @Stillwtm , I just tried this on mujoco and mujoco-mjx 3.3.0 and cannot reproduce the issue given the zip files. Can you try on these latest versions?

[image]

Stillwtm, 2025-04-01:
Thanks for the reply! I upgraded to `mujoco-mjx 3.3.0`, the issue persists. However, when I downgraded `jax` from `0.5.2` to `0.5.0`, the problem disappeared.

btaba, 2025-04-07:
Ok @Stillwtm this is pretty curious, I was able to reproduce the issue a couple of times, but not consistently even on jax==0.5.2.  Do you have a consistent way to repro the error?

Stillwtm, 2025-04-08:
@btaba I can consistently trigger this issue, but I'm not sure what causes it. Maybe I can provide my complete Python environment.

[environment.zip](https://github.com/user-attachments/files/19646291/environment.zip)