google-deepmind/mujoco #2487: Renderer API

westurner, 2025-03-09:
### The feature, motivation and pitch

- google/filament:
  - https://github.com/google/filament
  - Android/iOS/Lin/Mac/Win, WebGL
- LuisaGroup/LuisaRender:
  https://github.com/LuisaGroup/LuisaRender
  - Genesis-AI renders w/ LuisaRender
- https://mujoco.readthedocs.io/en/stable/programming/visualization.html
- https://mujoco.readthedocs.io/en/latest/unity.html
  - Also, Which other 3d platforms could Mujoco support:
    - https://github.com/panda3d/panda3d
    - https://github.com/godotengine/godot
    - https://github.com/the-mirror-gdp/the-mirror
    - https://github.com/jsdotlua/jest-lua

### Alternatives

- This is not a pressing need.
- I noticed your renderer, just learned about Filament (2018), and thought I'd share these opportunities.
- There could be mujoco_menagerie robots in MMORPG platforms and/or there could be high quality renders.

demoncoder-crypto, 2025-03-10:
Is this issue open i would love to contribute

yuvaltassa, 2025-03-11:
We are looking closely at Filament 🙂

demoncoder-crypto, 2025-03-13:
I will send a rough draft first on how to approach this. if it looks alright let me know

westurner, 2025-03-14:
To help define the problem:
And then overlay or composite 2D and 3D data visualizations with tests and/or Augmented Reality:
- TIL about https://datoviz.org/ which says it wraps glfw
  - Datoviz has Matplotlib and MATLAB colormaps
- wgpu is written in Rust and there are many bindings;
  - https://github.com/gfx-rs/wgpu
  - https://github.com/gfx-rs/wgpu-native#bindings
  - https://github.com/pygfx/pygfx wraps wgpu
    - glfw, jupyter_rfb (rendering in Jupyter), PyQt, PySide, or wx.

yuvaltassa, 2025-03-14:
`wgpu` looks quite interesting! thanks for sharing.

cc @saran-t