google-deepmind/mujoco #2489: NaNs in act when differentiating through muscle-driven arm26.xml with MJX

jamesheald, 2025-03-10:
### Intro

Hi!

I am a postdoc at University College London, I use MuJoCo for my research on musculoskeletal control.

I'm very excited by the prospect of using MJX to differentiate through the forward dynamics of musculoskeletal models. However, when I try to differentiate through mjx.step to calculate derivatives of muscle activations with respect to muscle inputs, I get NaNs. I have tested this on the [arm26](https://github.com/google-deepmind/mujoco/blob/main/mjx/mujoco/mjx/test_data/actuator/arm26.xml) model in the MJX Github repo. In contrast, when I calculate derivatives of qpos with respect to forces applied directly to the joints, I get well-behaved gradients. This suggests that there is a specific problem with calculating gradients through muscles.

### My setup

OS: MacOS Sequoia
python: 3.11.5
mujoco: 3.3.0
mujoco-mjx: 3.3.0
jax: 0.5.1
jaxlib: 0.5.1

### What's happening? What did you expect?

The script produces NaNs in the gradients of muscle activations with respect to muscle inputs, but not in the gradients of joint positions with respect to forces applied to the joints.

I expected to be able to apply jax.grad and jax.jacrev to mjx.step to calculate gradients of state variables with respect to muscle inputs.

### Steps for reproduction

Run the code below.

### Minimal model for reproduction

<details>
<summary>minimal XML</summary>

```XML

<mujoco model="2-link 6-muscle arm">
  <option timestep="0.005" iterations="50" solver="Newton" tolerance="1e-10"/>

  <visual>
    <rgba haze=".3 .3 .3 1"/>
  </visual>

  <default>
    <joint type="hinge" pos="0 0 0" axis="0 0 1" limited="true" range="0 120" damping="0.1"/>
    <muscle ctrllimited="true" ctrlrange="0 1"/>
  </default>

  <asset>
    <texture type="skybox" builtin="gradient" rgb1="0.6 0.6 0.6" rgb2="0 0 0" width="512" height="512"/>

    <texture name="texplane" type="2d" builtin="checker" rgb1=".25 .25 .25" rgb2=".3 .3 .3" width="512" height="512" mark="cross" markrgb=".8 .8 .8"/>

    <material name="matplane" reflectance="0.3" texture="texplane" texrepeat="1 1" texuniform="true"/>
  </asset>

  <worldbody>
    <geom name="floor" pos="0 0 -0.5" size="0 0 1" type="plane" material="matplane"/>

    <light directional="true" diffuse=".8 .8 .8" specular=".2 .2 .2" pos="0 0 5" dir="0 0 -1"/>

    <site name="s0" pos="-0.15 0 0" size="0.02"/>
    <site name="x0" pos="0 -0.15 0" size="0.02" rgba="0 .7 0 1" group="1"/>

    <body pos="0 0 0">
      <geom name="upper arm" type="capsule" size="0.045" fromto="0 0 0  0.5 0 0" rgba=".5 .1 .1 1"/>
      <joint name="shoulder"/>
      <geom name="shoulder" type="cylinder" pos="0 0 0" size=".1 .05" rgba=".5 .1 .8 .5" mass="0" group="1"/>

      <site name="s1" pos="0.15 0.06 0" size="0.02"/>
      <site name="s2" pos="0.15 -0.06 0" size="0.02"/>
      <site name="s3" pos="0.4 0.06 0" size="0.02"/>
[... 217 lines cut ...]
      mjx_data = mjx_data.replace(ctrl=ctrl)
      mjx_data = mjx.step(mjx_model, mjx_data)

      return mjx_data.act
    
    get_jacobian_muscles = jax.jit(jax.jacrev(muscle_step))

    # differentiation compatabile solver settings - https://github.com/google-deepmind/mujoco/issues/1182
    mjx_model = mjx_model.replace(opt=mjx_model.opt.replace(solver=mujoco.mjtSolver.mjSOL_NEWTON))
    mjx_model = mjx_model.replace(opt=mjx_model.opt.replace(iterations=1))
    mjx_model = mjx_model.replace(opt=mjx_model.opt.replace(ls_iterations=1))

    jacobian_joints = get_jacobian_joints(jp.ones(mjx_model.nq), mjx_data, mjx_model)

    if not jp.any(jp.isnan(jacobian_joints)): print("No NaNs detected when differentiating through joints")

    jacobian_muscles = get_jacobian_muscles(jp.ones(mjx_model.nu), mjx_data, mjx_model)

    assert not jp.any(jp.isnan(jacobian_muscles)), f"NaNs detected when differentiating through muscles"
```

### Confirmations

- [x] I searched the [latest documentation](https://mujoco.readthedocs.io/en/latest/overview.html) thoroughly before posting.
- [x] I searched previous [Issues](https://github.com/google-deepmind/mujoco/issues) and [Discussions](https://github.com/google-deepmind/mujoco/discussions), I am certain this has not been raised before.

jamesheald, 2025-03-10:
When I use mujoco.mjd_transitionFD to calculate the Jacobian in the muscle model, I get all zeros for the derivative of the next state with respect to the muscle inputs.

The Jacobian is (2*nv+na) x nu, where nv = 2 and nv and nu = 6: 
<pre>
Jacobian muscle model: 
 [[0. 0. 0. 0. 0. 0.]
 [0. 0. 0. 0. 0. 0.]
 [0. 0. 0. 0. 0. 0.]
 [0. 0. 0. 0. 0. 0.]
 [1. 0. 0. 0. 0. 0.]
 [0. 1. 0. 0. 0. 0.]
 [0. 0. 1. 0. 0. 0.]
 [0. 0. 0. 1. 0. 0.]
 [0. 0. 0. 0. 1. 0.]
 [0. 0. 0. 0. 0. 1.]]
</pre>

When I exchange the 6 muscles for a single hinge joint motor, I get all nonzeros for the derivative of the next state with respect to the motor input.

The values of nv and nu are now both 1: 
<pre>
Jacobian motor model: 
 [[-0.00012784]
 [ 0.00044407]
 [-0.02556708]
 [ 0.08881482]]
</pre>

This is the script I ran:

```python
import mujoco
import numpy as np

muscle_model_str = """
<mujoco model="2-link 6-muscle arm">
  <option timestep="0.005" iterations="50" solver="Newton" tolerance="1e-10"/>

  <visual>
    <rgba haze=".3 .3 .3 1"/>
  </visual>

  <default>
    <joint type="hinge" pos="0 0 0" axis="0 0 1" limited="true" range="0 120" damping="0.1"/>
    <muscle ctrllimited="true" ctrlrange="0 1"/>
  </default>

  <asset>
    <texture type="skybox" builtin="gradient" rgb1="0.6 0.6 0.6" rgb2="0 0 0" width="512" height="512"/>

    <texture name="texplane" type="2d" builtin="checker" rgb1=".25 .25 .25" rgb2=".3 .3 .3" width="512" height="512" mark="cross" markrgb=".8 .8 .8"/>

    <material name="matplane" reflectance="0.3" texture="texplane" texrepeat="1 1" texuniform="true"/>
  </asset>

  <worldbody>
    <geom name="floor" pos="0 0 -0.5" size="0 0 1" type="plane" material="matplane"/>

    <light directional="true" diffuse=".8 .8 .8" specular=".2 .2 .2" pos="0 0 5" dir="0 0 -1"/>

    <site name="s0" pos="-0.15 0 0" size="0.02"/>
    <site name="x0" pos="0 -0.15 0" size="0.02" rgba="0 .7 0 1" group="1"/>

    <body pos="0 0 0">
      <geom name="upper arm" type="capsule" size="0.045" fromto="0 0 0  0.5 0 0" rgba=".5 .1 .1 1"/>
      <joint name="shoulder"/>
      <geom name="shoulder" type="cylinder" pos="0 0 0" size=".1 .05" rgba=".5 .1 .8 .5" mass="0" group="1"/>

      <site name="s1" pos="0.15 0.06 0" size="0.02"/>
      <site name="s2" pos="0.15 -0.06 0" size="0.02"/>
[... 124 lines cut ...]
    <motor name="hinge" joint="elbow"/>
  </actuator>
</mujoco>
"""

if __name__ == "__main__":
    
    muscle_model = mujoco.MjModel.from_xml_string(muscle_model_str)
    data = mujoco.MjData(muscle_model)
    data.ctrl = np.ones(muscle_model.nu)
    B = np.zeros((2 * muscle_model.nv + muscle_model.na, muscle_model.nu))

    # Compute the transition Jacobian using finite differences
    mujoco.mjd_transitionFD(muscle_model, data, 1e-6, 1, None, B, None, None)
    print("Jacobian muscle model: \n", B)

    motor_model_str = mujoco.MjModel.from_xml_string(motor_model_str)
    data = mujoco.MjData(motor_model_str)
    data.ctrl = np.ones(motor_model_str.nu)
    B = np.zeros((2 * motor_model_str.nv + motor_model_str.na, motor_model_str.nu))

    # Compute the transition Jacobian using finite differences
    mujoco.mjd_transitionFD(motor_model_str, data, 1e-6, 1, None, B, None, None)
    print("Jacobian motor model: \n", B)
```

yuvaltassa, 2025-03-11:
This is working as intended and exactly what I would expect. When you have muscles, the excitation (ctrl) only affects the activation (act). The activation affects the dynamics through the transition matrix A, not through the control matrix B. ctrl -> act -> qacc -> qvel -> qpos.

Does this make sense?

jamesheald, 2025-03-11:
Maybe I have misunderstood the role of [transitionfd documentation](https://mujoco.readthedocs.io/en/stable/APIreference/APIfunctions.html#mjd-transitionfd). In terms of your flow (ctrl -> act -> qacc -> qvel -> qpos), I thought that B would contain dqpos/dctrl and dqvel/dctrl. How do I compute these quantities in regular MuJoCo?

More, generally I want to compute the derivatives of functions of qpos or qvel (e.g. the position and orientation of bodies) with respect to ctrl (ideally in MJX, but regular MuJoCo would also be useful).

yuvaltassa, 2025-03-11:
You did not misunderstand, that is exactly what it's giving you. In the model with the muscles dqpos/dctrl and dqvel/dctrl are **identically zero**, because ctrl only affects act (and then act affects qvel).

jamesheald, 2025-03-11:
Why are we not just applying the chain rule across all the step in your flow to get a nonzero dqpos/dctrl?

yuvaltassa, 2025-03-11:
This is a discrete time, stateful system. The `act` variables introduce a state **between** ctrl and the dynamical state. dqpos/dctrl **is zero** if you have muscles or other stateful actuators; the chain rule has nothing to do with it. That's what statefulness means, in discrete time. At time 0, `ctrl` will affect `act`, at the next timestep `act` will affect the dynamics.

jamesheald, 2025-03-11:
Ok, that makes sense. That doesn't explain why I get NaN when calculating dact/qctrl (my original post) though does it?

yuvaltassa, 2025-03-12:
No it does not. I'll reopen.

Balint-H, 2025-04-24:
I think I figured out the issue: none of the muscles were using tausmooth! Simply adding a smoothing value makes the gradients for dAct/dCtrl available! (otherwise you get nans down the line due to dividing by zero in `muscle_dynamics_timescale()`). 

Modifying the xml to the following, the nans disappear:
``` xml
<actuator>
    <muscle name="SF" tendon="SF" tausmooth="0.1"/>
    <muscle name="SE" tendon="SE" tausmooth="0.1"/>
    <muscle name="EF" tendon="EF" tausmooth="0.1"/>
    <muscle name="EE" tendon="EE" tausmooth="0.1"/>
    <muscle name="BF" tendon="BF" tausmooth="0.1"/>
    <muscle name="BE" tendon="BE" tausmooth="0.1"/>
  </actuator>
```

Alternatively you can set `actuator_dynprm[:, 2]` of your model.

closed 2025-04-24 by Balint-H

jamesheald, 2025-04-25:
Inspired by @Balint-H's fix above, I think I found a more general fix that works even with tausmooth = 0.

I've made a [PR](https://github.com/google-deepmind/mujoco/pull/2599). TLDR; I replaced jp.where with jax.lax.cond to avoid evaluating unused code branches that evaluate to NaN.