google-deepmind/mujoco #2491: Error: eigenvalue of mesh inertia must be positive: fl_knee

valeriaskvo, 2025-03-10:
### Intro

Hi!  

I am a PhD student at ITMO University, working on legged robots and using MuJoCo for my research. I am currently exploring different methods for dynamically modifying model parameters using MjSpec. My goal is to generate multiple MJCF models with slight variations to test different configurations for further training.  

To build my base model, I used the [`sw_urdf_exporter` script](https://wiki.ros.org/sw_urdf_exporter), first generating a URDF model, then converting it to XML and manually adding additional elements.

Currently, I am trying to dynamically add motor parameters and sensors to my model to determine which configurations work best for my robot’s geometry. However, I encountered an issue during the model generation process using MjSpec.  

I would appreciate any help or insights on this!  

### My setup

I am using `mujoco==3.2.7`, python API. The bug occurs on both Ubuntu 22 and Windows OS.  

I previously reported an [issue](https://github.com/google-deepmind/mujoco_playground/issues/74) with the same problem in the MuJoCo Playground, but changing the library version from `3.3.0` to `3.2.7` helped in solving this problem

### What's happening? What did you expect?

I ran `spec.compile()` multiple times. On the second run, I encountered the following error on Windows OS:  
`ValueError: Error: eigenvalue of mesh inertia must be positive: fl_knee`  

And the following error on Linux OS:
`ValueError: Error: eigenvalue of mesh inertia must be positive: base`

Could you help me understand what is wrong with my model? Why does this error occur with my model but not with the ones available in [mujoco_menagerie](https://github.com/google-deepmind/mujoco_menagerie/tree/main)?

### Steps for reproduction

To reproduce the bug, you can use this [Google Colab example](https://colab.research.google.com/drive/1OsIzv-KWul2JLdHwhNroZUYOfG3OdbGa?usp=sharing), just run all cells.  

### Minimal model for reproduction

My model, called **Cheetah**, can be found in this repository: [cheetah_mujoco](https://github.com/valeriaskvo/cheetah_mujoco.git), specifically in this file: [cheetah.xml](https://github.com/valeriaskvo/cheetah_mujoco/blob/main/cheetah/cheetah.xml).  

### Code required for reproduction

```python
import mujoco

filepath = 'cheetah_mujoco/cheetah/cheetah.xml'
spec = mujoco.MjSpec.from_file(filepath)
print("Compile 1")
spec.compile()
print("Compile 2")
spec.compile()
print("Compile 3")
spec.compile()
print("Compile 4")
spec.compile()
print("Compile 5")
spec.compile()
```

### Confirmations

- [x] I searched the [latest documentation](https://mujoco.readthedocs.io/en/latest/overview.html) thoroughly before posting.
- [x] I searched previous [Issues](https://github.com/google-deepmind/mujoco/issues) and [Discussions](https://github.com/google-deepmind/mujoco/discussions), I am certain this has not been raised before.

valeriaskvo, 2025-03-10:
Thank you very much for the quick response.  

I managed to build MuJoCo 3.3.1, but I was unable to build the Python API version. I followed [the instructions](https://mujoco.readthedocs.io/en/latest/python.html#building-from-source), but I am not sure if I correctly built the binary release.  

When trying to install, I encounter an error related to paths:  
```
subprocess.CalledProcessError: Command '['cmake', '-DPython3_ROOT_DIR:PATH=/home/user/cheetah/cheetah_mujoco/mujoco/python/venv', '-DPython3_EXECUTABLE:STRING=/home/user/cheetah/cheetah_mujoco/mujoco/python/venv/bin/python', '-DCMAKE_MODULE_PATH:PATH=/tmp/pip-req-build-5t4ns9bc/mujoco/cmake', '-DCMAKE_BUILD_TYPE:STRING=Release', '-DCMAKE_LIBRARY_OUTPUT_DIRECTORY:PATH=build/temp.linux-x86_64-cpython-39', '-DCMAKE_INTERPROCEDURAL_OPTIMIZATION:BOOL=ON', '-DCMAKE_Fortran_COMPILER:STRING=', '-DBUILD_TESTING:BOOL=OFF', '-DMUJOCO_LIBRARY_DIR:PATH=/home/user/cheetah/cheetah_mujoco/mujoco_build', '-DMUJOCO_INCLUDE_DIR:PATH=/home/user/cheetah/cheetah_mujoco/mujoco/include/mujoco', '-DPython3_LIBRARY=/usr/local/lib/python3.9', '-DPython3_INCLUDE_DIR=/usr/local/include/python3.9', '/tmp/pip-req-build-5t4ns9bc/mujoco']' returned non-zero exit status 1.
```
Could you please advise on the correct way to build the version so that it works with Python before your official release?

valeriaskvo, 2025-03-10:
Also I have a question regarding the alternative method **"don't set the density to 0 in the defaults"**.  

How can I disable inertia calculation based on geometry? I have explicitly set the inertia parameters for all bodies (or I will set them manually).  

I tried using:  
```
<compiler inertiafromgeom="false"/>
```  
but it didn’t work.  

How can I achieve this?

quagla, 2025-03-10:
I don't know how to answer the install question..

Concerning the other question, the mesh processing is performed in any case, even if you set `inertiafromgeom` to false. But the default is `auto`, so if you set the `inertial` it's already equivalent to `false`. Is there a specific reason why you want to bypass the mesh processing completely?

valeriaskvo, 2025-03-10:
I don't understand the nature of the error I'm having. I understand that I have the inertia and mass of the model set, I don't quite understand how to analyze at the model generation stage whether I have a problem with eigenvalues. I have tried reproducing the error using the inertia calculation via the `trimesh` library, however no problems occurred.

I have also worked with a lot of custom models, and this is the first time I have encountered such an error. Do you have any recommendations on how to validate .stl files before integrating them into a model?

Translated with DeepL.com (free version)

valeriaskvo, 2025-03-11:
Your response has helped me a lot. The problem has been solved, I am waiting for the release of version 3.3.1. Thanks!

closed 2025-03-11 by valeriaskvo