google-deepmind/mujoco #2494: Exposing handles for plotting, image overlay and text boxes from simulate.cc and viewer.py

Andrew-Luo1, 2025-03-11:
### The feature, motivation and pitch

My RL workflow is Brax for its PPO, MJX for physics, and Madrona-MJX rendering for pixel observations. Especially with the detailed reward shaping for manipulation tasks and the visual domain-randomization required for sim2real transfer of pixel-based policies, visualizing rollouts throughout the training process across different envs is very helpful.

Late last year, I wrote *rscope*, a fork of Mujoco's viewer for visualizing RL training. It has become a key part of my workflow and I'd love to open source it.

https://github.com/user-attachments/assets/2785e4cc-93a8-4835-afe0-bac832d0e6cc

Rscope allows the user to inspect Brax training runs at no additional cost. Brax already punctuates the training run with *evaluator rollouts* for policy evaluation. Rscope visualizes the transitions generated from these rollouts.

**Mujoco Change Proposal**
To cleanly release rscope, I would like to implement three general mujoco viewer features that have previously been requested by other users: text boxes, plotting and image buffer overlay. I believe the following changes would be required:

`mujoco/simulate/simulate.cc`
Methods to initialize, draw and toggle the proposed assets.

`mujoco/python/mujoco/simulate.cc`
Python bindings for the new methods.

`mujoco/python/mujoco/viewer.py`
Expose the python bindings by modifying `class Handle`, which is returned by the passive viewer.

The required changes should be quite minor (<300 lines). No build changes should be needed, since these new features would simply use API's already called elsewhere in the files.

### Alternatives

**Plan B**. An alternative that would require less or no changes to Mujoco would be to essentially re-implement a simpler version of `mujoco/simulate/simulate.cc` in Python via the visualization bindings. [This repository](https://github.com/rohanpsingh/mujoco-python-viewer) provides an example of that.

While this is possible, I'm currently of the opinion that Plan A is the way to go. Plan B would likely require either importing that github repo as an additional dependency or adding ~900 lines of code. Additionally, the three proposed features can be useful to the many existing users of the passive viewer.

yuvaltassa, 2025-03-11:
This is a great proposal! 🚀 

One question is, do we first need to add the `mjui` structs to `python/mujoco/introspect` ?

Also, I believe @josechenf and @silviacruciani have worked on something like this internally, perhaps they have some opinions.

vikashplus, 2025-03-12:
My motivation for [this request](https://github.com/google-deepmind/mujoco/issues/2362) was somewhat similar. It will allow me to bring plots/visuals even from sources outside of MuJoCo

silviacruciani, 2025-03-12:
This matches the work that was done internally for exposing adding plots to the viewer
(see commit 37d7591ce0d1424f6853dd9293eec1579c8e623f) but proposes adding even more functionality, which is great 🚀

yuvaltassa, 2025-03-13:
@Andrew-Luo1, sounds like you have a green light for plan A, go for it!  🙂 

I still think we can't do custom UI without the mjUI bindings, but this would definitely be a good start.