google-deepmind/mujoco #2505: mjpython on MacOS does not support other 3rd party packages

Fima2003, 2025-03-17:
### Intro

Hi!

I am an undergrad student at Technion, I use MuJoCo for my research on Robotic arms.

### My setup

I am using MuJoCo on python on Mac(M-series chip) with Sequoia 15.3.2, and I cannot import not pre-built packages when running with mjpython

### What's happening? What did you expect?

Well, I expected it to run, but it does

### Steps for reproduction

1. Load the package below.
`pip3 install mink`
2. Run the code below.(with mjpython setup.py)
```python
import mujoco
import mink
import numpy as np

def setup():
    XML_PATH = "ur5e_scene.xml"
    model = mujoco.MjModel.from_xml_path(XML_PATH)
    data = mujoco.MjData(model)

    key_name = "home"
    key_id = mujoco.mj_name2id(model, mujoco.mjtObj.mjOBJ_KEY, key_name)
    if key_id >= 0:
        data.qpos[:] = model.key_qpos[key_id].copy()
        data.ctrl[:] = model.key_ctrl[key_id].copy()

    # Make sure the state is updated:
    mujoco.mj_forward(model, data)

    return model, data

if __name__ == "__main__":
    setup()
```    
3. See the error 
```
Traceback (most recent call last):
  File "/universal_robots_ur5e/setup.py", line 2, in <module>
    import mink
ModuleNotFoundError: No module named 'mink'
```

### Minimal model for reproduction

The model is not that relevant, to be honest, but for argument's sake, you can use the default one:

<details>
<summary>minimal XML</summary>

```XML

<mujoco>
  <worldbody>
    <light pos="0 0 1"/>
    <geom type="sphere" size="1" rgba="1 0 0 1"/>
  </worldbody>
</mujoco>

```
</details>

### Code required for reproduction

```
import mujoco
import mink
import numpy as np

def setup():
    XML_PATH = "ur5e_scene.xml"
    model = mujoco.MjModel.from_xml_path(XML_PATH)
    data = mujoco.MjData(model)

    key_name = "home"
    key_id = mujoco.mj_name2id(model, mujoco.mjtObj.mjOBJ_KEY, key_name)
    if key_id >= 0:
        data.qpos[:] = model.key_qpos[key_id].copy()
        data.ctrl[:] = model.key_ctrl[key_id].copy()

    # Make sure the state is updated:
    mujoco.mj_forward(model, data)

    return model, data

if __name__ == "__main__":
    setup()
```

### Confirmations

- [x] I searched the [latest documentation](https://mujoco.readthedocs.io/en/latest/overview.html) thoroughly before posting.
- [x] I searched previous [Issues](https://github.com/google-deepmind/mujoco/issues) and [Discussions](https://github.com/google-deepmind/mujoco/discussions), I am certain this has not been raised before.

Fima2003, 2025-03-17:
@yuvaltassa this is not mink related issue as the rest of the 3rd party packages are also not loading. Should have mentioned that in the issue

yunho-c, 2025-08-02:
This may be because `mjpython` is pointing to a Python that does not have `mink` installed (i.e., not the same one that `pip` is pointing to).