google-deepmind/mujoco #2508: Inconsistent Behavior qLD Mujoco & MJX

MoritzSchwerdt, 2025-03-17:
### Intro

Hi!

I am a undergrad student at Goethe University & TU Darmstadt, I use MuJoCo for my research on Physics 3D BinPacking

### My setup

Mujoco 3..3.0

### What's happening? What did you expect?

Mujoco and MJX seem to handle qLD differently in stepping, leading to Value Error when retrieving mj_data like

"mj_data = mjx.get_data(mj_model, mjx_data)"

### Steps for reproduction

1. Load the model below.
2. Run the code below.
3. Observe Value Error in qLD after mjx.step

### Minimal model for reproduction

1. Define Mujoco with 15+ Boxes

xml = "<mujoco> <worldbody>"
for i in range(15):
  xml += f"""
    <body name="box{i}">
      <joint name="box_{i}" type="free"/>
      <geom name="box_geom_{i}" type="box" size=".2 .2 .2" rgba="1 0 0 1" condim="3"/>
    </body>
  """
xml += "</worldbody></mujoco>"

2. Load into Mujoco & MJX
mj_model = mujoco.MjModel.from_xml_string(xml)
mj_data = mujoco.MjData(mj_model)

mjx_model = mjx.put_model(mj_model)
mjx_data = mjx.put_data(mj_model, mj_data)

3. Step & observe Error

stepped_mjx_data = mjx.step(mjx_model, mjx_data)
mjx_data.qLD.shape

stepped_mj_data = mjx.get_data(mj_model, stepped_mjx_data)

Leading to:
ValueError: Input field qLD has shape (315,), but output has shape (90,)

### Code required for reproduction

```python
import mujoco
from mujoco import mjx

xml = "<mujoco> <worldbody>"
for i in range(15):
  xml += f"""
    <body name="box{i}">
      <joint name="box_{i}" type="free"/>
      <geom name="box_geom_{i}" type="box" size=".2 .2 .2" rgba="1 0 0 1" condim="3"/>
    </body>
  """
xml += "</worldbody></mujoco>"

# Make model, data
mj_model = mujoco.MjModel.from_xml_string(xml)
mj_data = mujoco.MjData(mj_model)

# put onto GPU
mjx_model = mjx.put_model(mj_model)
mjx_data = mjx.put_data(mj_model, mj_data)

# Step Model
stepped_mjx_data = mjx.step(mjx_model, mjx_data)

# Obtain Updated Data
mj_data = mjx.get_data(mj_model, stepped_mjx_data)
```

### Confirmations

- [x] I searched the [latest documentation](https://mujoco.readthedocs.io/en/latest/overview.html) thoroughly before posting.
- [x] I searched previous [Issues](https://github.com/google-deepmind/mujoco/issues) and [Discussions](https://github.com/google-deepmind/mujoco/discussions), I am certain this has not been raised before.

yuvaltassa, 2025-03-18:
Nice catch. 
This will be fixed shortly.

btaba, 2025-04-07:
Looks like it's fixed, closing the issue!

hzheng40, 2025-09-30:
This issue still exists (with @MoritzSchwerdt 's code) on version 3.3.6 with python3.13 and jax 0.7.0.

hzheng40, 2025-09-30:
Looks like it was fixed in 3.3.1 and 3.3.2 but popped back up in 3.3.3

hzheng40, 2025-10-01:
Yes, also in 3.3.6 @yuvaltassa

btaba, 2025-10-03:
Fix incoming