google-deepmind/mujoco #2541: [MJX] Body with freejonit occur pjit nan error

rivergold, 2025-03-28:
### Intro

Hi!

I am a student at UESTC, I use MuJoCo for my research on robot control. I want to use MJX and madrona to do RL.

### My setup

MuJoco version: 3.3.0
OS: Ubuntu 22.04
GPU: NVIDIA GeForce RTX 3090
Driver: 565.77
Cuda version: 12.5

### What's happening? What did you expect?

When run with CPU use `simulate example_mjx.xml` everything is ok. But when use `mjx-viewer --mjcf example_mjx.xml` will occur error

```
dx = step_fn(mx, dx)
...
jax._src.dispatch.InternalFloatingPointError: ('pjit', 'nan')
```

I know that Jax JIT is success, the error occurs when do step, but I dont know how to solve this. I search this [issue](https://github.com/google-deepmind/mujoco/issues/1484) is relevant, but it cannot help to solve the error.

When I delete the freejoint on the object, it can run, but I must use the freejoint!

And I want to explain I use multi box type geoms as collision:
1. I use [obj2mjcf](https://github.com/kevinzakka/obj2mjcf) (with [CoACD](https://github.com/SarahWeiii/CoACD)) to generate mjcf and collision
2. Multi mesh files run very slow with mjx, and I refences this [issue](https://github.com/google-deepmind/mujoco/issues/1273#issuecomment-1883609335) to convert mesh into box type geom, the JIT time is decrease.
3. But it occur errors……

How to solve this problem? What did I do wrong? This error blocks all my work，I kindly request your prompt assistance.

### Steps for reproduction

1. Load the model below.
2. Run `mjx-viewer --mjcf example_mjx.xml`

### Minimal model for reproduction

I upload the whole xml with assets [report_issue.zip](https://github.com/user-attachments/files/19501845/report_issue.zip)

```xml
<?xml version="1.0" ?>
<mujoco model="empty_arena">
    <compiler angle="radian" assetdir="assets/"/>

    <visual>
        <global azimuth="120" elevation="-20"/>
        <headlight ambient="0.3 0.3 0.3" diffuse="0.6 0.6 0.6" specular="0 0 0"/>
        <rgba haze="0.15 0.25 0.35 1"/>
    </visual>

    <statistic extent="1.2" center="0 0 0.3"/>

    <asset>
        <texture type="skybox" builtin="gradient" rgb1="0.3 0.5 0.7" rgb2="0 0 0" width="512" height="3072"/>
        <texture type="2d" name="groundplane" builtin="checker" mark="edge" rgb1="0.2 0.3 0.4" rgb2="0.1 0.2 0.3" markrgb="0.8 0.8 0.8" width="300" height="300"/>
        <texture type="2d" name="fixture--workbench--workbench_0_fixture--workbench--workbench_0@work_bench_Roughness" file="work_bench_Roughness.png"/>
        <material name="groundplane" texture="groundplane" texuniform="true" texrepeat="5 5" reflectance="0.2"/>
        <material name="fixture--workbench--workbench_0_fixture--workbench--workbench_0@work_bench" texture="fixture--workbench--workbench_0_fixture--workbench--workbench_0@work_bench_Roughness" specular="0" shininess="0.02"/>
        <material name="object--box--box_0_object--box--box_0@Dirt_PVC" rgba="0.050328 0.112373 0.800165 1"/>
        <mesh name="fixture--workbench--workbench_0_fixture--workbench--workbench_0" file="fixture--workbench--workbench_0.obj"/>
        <mesh name="object--box--box_0_object--box--box_0" file="object--box--box_0.obj"/>
    </asset>

    <worldbody>
        <geom name="floor" size="0 0 0.05" type="plane" group="1" material="groundplane"/>
[... 62 lines cut ...]
                <geom name="object--box--box_0_collision_48" size="0.0298649 0.00482435 0.0559611" pos="-0.190611 0.0358383 -0.0254779" quat="0.495556 -0.495557 -0.504404 -0.504404" type="box" friction="0.95 0.3 0.1" solref="0.001" solimp="0.998 0.998" density="100" rgba="0 0.5 1 0.5"/>
                <geom name="object--box--box_0_collision_49" size="0.00303931 0.0869834 0.012355" pos="-0.0866982 -0.132097 -0.027452" quat="0.493011 -0.506893 -0.493011 -0.506893" type="box" friction="0.95 0.3 0.1" solref="0.001" solimp="0.998 0.998" density="100" rgba="0 0.5 1 0.5"/>
                <geom name="object--box--box_0_collision_50" size="0.0342892 0.0153087 0.00623985" pos="-0.192606 -0.0544121 -0.0109902" quat="0.500013 -0.500013 0.499987 -0.499987" type="box" friction="0.95 0.3 0.1" solref="0.001" solimp="0.998 0.998" density="100" rgba="0 0.5 1 0.5"/>
                <geom name="object--box--box_0_collision_51" size="0.00931835 0.00822707 0.0342891" pos="-0.195683 -0.0544119 -0.0345276" quat="5.33851e-08 -5.33851e-08 -0.707107 0.707107" type="box" friction="0.95 0.3 0.1" solref="0.001" solimp="0.998 0.998" density="100" rgba="0 0.5 1 0.5"/>
                <geom name="object--box--box_0_collision_52" size="0.0179866 0.00771944 0.01573" pos="-0.181652 -0.106247 -0.0114114" quat="5.48457e-08 0.726454 -0.687215 -5.18833e-08" type="box" friction="0.95 0.3 0.1" solref="0.001" solimp="0.998 0.998" density="100" rgba="0 0.5 1 0.5"/>
                <geom name="object--box--box_0_collision_53" size="0.00780662 0.0297791 0.0115139" pos="-0.187689 -0.113545 -0.0349481" quat="0.150204 -0.690969 -0.150204 -0.690969" type="box" friction="0.95 0.3 0.1" solref="0.001" solimp="0.998 0.998" density="100" rgba="0 0.5 1 0.5"/>
                <geom name="object--box--box_0_collision_54" size="0.0477261 0.0126555 0.00399347" pos="-0.194698 -0.0671013 -0.0467482" quat="5.57813e-08 -0.673875 -0.738845 -5.08762e-08" type="box" friction="0.95 0.3 0.1" solref="0.001" solimp="0.998 0.998" density="100" rgba="0 0.5 1 0.5"/>
                <geom name="object--box--box_0_collision_55" size="0.0137285 0.00207672 0.0156181" pos="-0.197151 0.0747739 -0.0112995" quat="5.33851e-08 0.707107 -0.707107 -5.33851e-08" type="box" friction="0.95 0.3 0.1" solref="0.001" solimp="0.998 0.998" density="100" rgba="0 0.5 1 0.5"/>
                <geom name="object--box--box_0_collision_56" size="0.00496392 0.0559611 0.00952726" pos="-0.200039 0.0358383 -0.0364449" type="box" friction="0.95 0.3 0.1" solref="0.001" solimp="0.998 0.998" density="100" rgba="0 0.5 1 0.5"/>
                <geom name="object--box--box_0_collision_57" size="0.0232588 0.00805242 0.0061314" pos="-0.151027 -0.127797 -0.0366233" quat="7.53923e-08 0.998601 -0.0528757 -3.992e-09" type="box" friction="0.95 0.3 0.1" solref="0.001" solimp="0.998 0.998" density="100" rgba="0 0.5 1 0.5"/>
                <geom name="object--box--box_0_collision_58" size="0.0366174 0.00668238 0.00301703" pos="-0.185626 -0.0567349 -0.0557057" quat="0.650984 0.27501 0.275335 0.651752" type="box" friction="0.95 0.3 0.1" solref="0.001" solimp="0.998 0.998" density="100" rgba="0 0.5 1 0.5"/>
                <geom name="object--box--box_0_collision_59" size="0.0164842 0.00790984 0.00382514" pos="-0.17896 -0.10565 -0.0545683" quat="6.81616e-08 0.902827 -0.430004 -3.24644e-08" type="box" friction="0.95 0.3 0.1" solref="0.001" solimp="0.998 0.998" density="100" rgba="0 0.5 1 0.5"/>
                <geom name="object--box--box_0_collision_60" size="0.0640192 0.00211404 0.0061314" pos="-0.0639894 -0.124258 -0.0366233" quat="7.54979e-08 0 -1 0" type="box" friction="0.95 0.3 0.1" solref="0.001" solimp="0.998 0.998" density="100" rgba="0 0.5 1 0.5"/>
                <geom name="object--box--box_0_collision_61" size="0.00230218 0.00507107 0.0061314" pos="-0.0447315 -0.131445 -0.0366233" quat="7.54979e-08 0 -1 0" type="box" friction="0.95 0.3 0.1" solref="0.001" solimp="0.998 0.998" density="100" rgba="0 0.5 1 0.5"/>
                <site name="object--box--box_0_default_site" pos="0 0 0" size="0.002" rgba="1 0 0 1"/>
            </body>
        </body>
    </worldbody>
</mujoco>
```

### Confirmations

- [x] I searched the [latest documentation](https://mujoco.readthedocs.io/en/latest/overview.html) thoroughly before posting.
- [x] I searched previous [Issues](https://github.com/google-deepmind/mujoco/issues) and [Discussions](https://github.com/google-deepmind/mujoco/discussions), I am certain this has not been raised before.

btaba, 2025-04-07:
Hi @rivergold, add:

```
    <option timestep="0.001" iterations="20" ls_iterations="20"/>

    <default>
     <mesh maxhullvert="32"/>
    </default>
```
to get rid of the NaN. Feel free to tune the environment further for speed.

closed 2025-04-07 by btaba

rivergold, 2025-04-08:
@btaba 
Thank you for your response. However, the issue still persists after add `option` and `maxhullvert`

rivergold, 2025-04-08:
This issue has been troubling me for quite some time, and I'm struggling to make progress because of it.
I want to know why this simple env cannot run with mjx on GPU, and how to solve this?
I think use [obj2mjcf](https://github.com/kevinzakka/obj2mjcf) (with [CoACD](https://github.com/SarahWeiii/CoACD)) is a good way to build custom env for mujoco. But this simple env is failed. So sad......

btaba, 2025-04-08:
Hi @rivergold , I cannot repro the issue with the aforementioned changes with the latest jax/mujoco/mujoco-mjx. I'm on an A5000.

We've seen that some devices have numerical precision issues, here are a few suggestions:

```
from jax import config

# Switching to f64 kills perf, but would tell you if it's a numerical precision issue:
config.update("jax_enable_x64", True)
# this is a lighter weight way to resolve some precision issues on the GPU:
config.update('jax_default_matmul_precision', jax.lax.Precision.HIGH)

# config.update("jax_debug_nans", True) # if you want to dive deeper
```

btaba, 2025-04-08:
Keep in mind that mesh collisions are not performant in MJX at the moment, as per [this doc](https://mujoco.readthedocs.io/en/stable/mjx.html#mjx-the-sharp-bits). This is being resolved via https://github.com/google-deepmind/mujoco_warp. But that is irrespective of the NaN.

rivergold, 2025-04-09:
Thanks a lot. I tried all methods.

- RTX3090: Setting `jax_default_matmul_precision` to `high` solves the error
- A100: Setting `jax_default_matmul_precision` to `highest` solves the error
Ref [jax issue 18934](https://github.com/jax-ml/jax/issues/18934)
- Add `config.update("jax_debug_nans", True)`, the error message is as same as  `False`, ref [Jax doc](https://docs.jax.dev/en/latest/debugging/flags.html). This config doesn't work for pjit.

But with `config.update("jax_enable_x64", True)`

Errors occur as follows:

```
  File "/root/software/anaconda/lib/python3.11/site-packages/jax/_src/interpreters/pxla.py", line 3231, in check_arg_avals_for_call
    raise TypeError(
TypeError: Argument types differ from the types for which this computation was compiled. The mismatches are:
Argument 'd.contact.geom1' compiled with int32[496] and called with int64[496]
Argument 'd.contact.geom2' compiled with int32[496] and called with int64[496]
Argument 'd.contact.geom' compiled with int32[496,2] and called with int64[496,2]
```

I searched for an [issue #2237](https://github.com/google-deepmind/mujoco/issues/2237) about this error. I think this bug is already fixed in Mujoco/MJx=3.3.0. Could you please take another look for me? Sorry to trouble you.

> Hi [@rivergold](https://github.com/rivergold) , I cannot repro the issue with the aforementioned changes with the latest jax/mujoco/mujoco-mjx. I'm on an A5000.
>
> We've seen that some devices have [...]

rivergold, 2025-04-09:
I know that the function `make_data` fixes the bug, but `put_data` does not, and the bug was fixed yesterday. Nice.

btaba, 2025-04-09:
Hi @rivergold , thanks for the update! Would you mind creating a new bug for the new issue? Would really appreciate it. I think the fix is pretty straightforward, but I just want to make sure I'm fixing the bug you're seeing.

rivergold, 2025-04-10:
@btaba A new issue #2565  is created. Thanks for your recent support.