google-deepmind/mujoco #2564: Expose MjSpec raw pointer from python

tawfiq-jawhar, 2025-04-09:
### The feature, motivation and pitch

Python bindings has wrappers around the raw mujoco pointers and does not expose the raw pointers. Exposing the raw pointer in the binding will give flexibility to design solutions where a python procedural modelling api with mjspec is used with  C++ sim applications.  

1. Expose the raw pointer from the python wrapper of MjSpec, MjModel and MjData. 
2. Allow to create MjSpec from raw pointer in python. (expose the constructor `MjSpec(raw::MjSpec* ptr, const py::dict& assets_ = {});`)

quagla, 2025-04-09:
MjSpec has the attribute `_address`, I think that's what you want?

quagla, 2025-04-09:
However, in a C++ application, why not using the C API directly?

tawfiq-jawhar, 2025-04-09:
Thanks @quagla I didn't know about `_address` this should work to pass this pointer to C++. 
Exposing the `MjSpec(raw::MjSpec* ptr, const py::dict& assets_ = {});` constructor would be nice to move the other way too. 

I already have a python api that uses MjSpec that I want to use in Python and C++ to build up the spec.

quagla, 2025-04-10:
Would you like to send us a PR adding the constructor that you proposed?

closed 2025-07-14 by quagla