google-deepmind/mujoco #2572: [UsdExporter] Length scales are tiny in exported USD

siddancha, 2025-04-12:
### Intro

Hi!

I am a researcher at MIT, I use MuJoCo for my research on robotics.

### My setup

Ubuntu 22.04, MuJoCo v3.3.0, Python API

### What's happening? What did you expect?

When I export my model to USD and visualize it in Blender and Omniverse, the model is tiny! I need to scroll in a lot to be able to see my scene. Here's what I see in Omniverse:

https://github.com/user-attachments/assets/b49328a6-942b-4052-9419-9ad9cbd0809a

### Steps for reproduction

Here's my code snippet to reproduce this:

```python
import mujoco
import mujoco.viewer
from mujoco.usd import exporter

# Load the model -- Franka Emika Panda from Menagerie
model = mujoco.MjModel.from_xml_path("scene.xml")
data = mujoco.MjData(model)

# Create USD exporter
usd_exp = exporter.USDExporter(model=model)

duration = 5
framerate = 60

# Initialize the viewer
with mujoco.viewer.launch_passive(model, data) as viewer:
    # Simulation loop
    while viewer.is_running() and data.time < duration:
        # Step the simulation
        mujoco.mj_step(model, data)
        
        if usd_exp.frame_count < data.time * framerate:
            # Update the USD scene with current state
            usd_exp.update_scene(data=data)
        
        # Update the viewer
        viewer.sync()

# Save the animation to USD file
usd_exp.save_scene(filetype="usd")
```

### Minimal model for reproduction

My `scene.xml` copied from Menagerie (Franka Emika Panda's scene.xml)

```XML

<mujoco model="panda scene">
  <include file="panda.xml"/>

  <statistic center="0.3 0 0.4" extent="1"/>

  <visual>
    <headlight diffuse="0.6 0.6 0.6" ambient="0.3 0.3 0.3" specular="0 0 0"/>
    <rgba haze="0.15 0.25 0.35 1"/>
    <global azimuth="120" elevation="-20"/>
  </visual>

  <asset>
    <texture type="skybox" builtin="gradient" rgb1="0.3 0.5 0.7" rgb2="0 0 0" width="512" height="3072"/>
    <texture type="2d" name="groundplane" builtin="checker" mark="edge" rgb1="0.2 0.3 0.4" rgb2="0.1 0.2 0.3"
      markrgb="0.8 0.8 0.8" width="300" height="300"/>
    <material name="groundplane" texture="groundplane" texuniform="true" texrepeat="5 5" reflectance="0.2"/>
  </asset>

  <worldbody>
    <light pos="0 0 1.5" dir="0 0 -1" directional="true"/>
    <geom name="floor" size="0 0 0.05" type="plane" material="groundplane"/>
    <camera name="main_camera" pos="1.5 1.5 1.5" quat="0.5 -0.5 0.5 0.5"/>
  </worldbody>
</mujoco>

```
</details>

### Code required for reproduction

```python
import mujoco
import mujoco.viewer
from mujoco.usd import exporter

# Load the model -- Franka Emika Panda from Menagerie
model = mujoco.MjModel.from_xml_path("scene.xml")
data = mujoco.MjData(model)

# Create USD exporter
usd_exp = exporter.USDExporter(model=model)

duration = 5
framerate = 60

# Initialize the viewer
with mujoco.viewer.launch_passive(model, data) as viewer:
    # Simulation loop
    while viewer.is_running() and data.time < duration:
        # Step the simulation
        mujoco.mj_step(model, data)
        
        if usd_exp.frame_count < data.time * framerate:
            # Update the USD scene with current state
            usd_exp.update_scene(data=data)
        
        # Update the viewer
        viewer.sync()

# Save the animation to USD file
usd_exp.save_scene(filetype="usd")
```

### Confirmations

- [x] I searched the [latest documentation](https://mujoco.readthedocs.io/en/latest/overview.html) thoroughly before posting.
- [x] I searched previous [Issues](https://github.com/google-deepmind/mujoco/issues) and [Discussions](https://github.com/google-deepmind/mujoco/discussions), I am certain this has not been raised before.

abhihjoshi, 2025-04-28:
It seems you are using a plane of size 300x300. I am guessing this is very large relative to the rest of the scene which might be why you have to zoom in a lot?

siddancha, 2025-04-28:
@abhihjoshi Even if the plane is huge, I still think the dimensions are off. Each blue Mujoco square is of the order of 1m x 1m. And the white gridlines of Omniverse's default grid that you see after zooming out towards the end of the video should also be around 1m x 1m wide. So it does seem like the dimensions are being shrunk.

Suessydo, 2025-06-19:
Hello! Are you still following this post? I'm sorry I can't solve your problem. I used a code similar to yours, but the exported USD file disappears in Omniverse after **five seconds of running**.
Have you encountered a similar situation? Have you solved this problem?
My code:

```
def mjcf2usd(mjcf_path):
    # Load the model -- Franka Emika Panda from Menagerie
    model = mujoco.MjModel.from_xml_path(mjcf_path)
    data = mujoco.MjData(model)

    # Create USD exporter
    usd_exp = exporter.USDExporter(model=model)

    duration = 5
    framerate = 60

    # Initialize the viewer
    with mujoco.viewer.launch_passive(model, data) as viewer:
        # Simulation loop
        while viewer.is_running() and data.time < duration:
            # Step the simulation
            mujoco.mj_step(model, data)
            
            if usd_exp.frame_count < data.time * framerate:
                # Update the USD scene with current state
                usd_exp.update_scene(data=data)
            
            # Update the viewer
            viewer.sync()

    # Save the animation to USD file
    usd_exp.save_scene(filetype="usd")
```
I think the reason for this is duration=5, but I haven't found a solution for a long time.

yuvaltassa, 2025-07-28:
[MuJoCo doesn't have units](https://mujoco.readthedocs.io/en/latest/overview.html#units-are-unspecified), so I think this needs to be fixed on the USD side?

In any case, you should take a look at https://github.com/google-deepmind/mujoco/tree/main/src/experimental/usd, this will soon replace the current USD exporter. I'll let you know here when it's out of experimentall

havess, 2026-02-07:
Hi @siddancha !

At the moment your best bet is still (1). The new exporter is still missing Python bindings at the moment as I figure out how we're going to do distribution but it should be available in a release soon with accompanying tutorials.

For (3) that is a utility that will convert MJCF to USD assets that are meant to be roundtrip-able back into MuJoCo.