google-deepmind/mujoco #2580: The problem of the elongation of the tendon around a fixed pulley

tttj12, 2025-04-16:
### Intro

Hi!

I am a student ， I use MuJoCo for my research on Rope-driven manipulator.

### My setup

mujoco3.2.5

### What's happening? What did you expect?

We attempted to use tendons to drive the manipulator in order to simulate the rope-driven manipulator in reality. Therefore, the elongation of the tendon is of great importance to us. We found that when the tendon goes around a cylinder, the length of the tendon, whether it is the visual length or the simulated length, is calculated as a straight line. Could you please tell us what methods are available to improve the calculation accuracy so that the tendon will not pass through the geom?

### Steps for reproduction

1. Load the model below.
2. Run the code below.

### Minimal model for reproduction

<mujoco>
  <default >
    <joint armature="1e-05" damping="0.05"/>
    <site size="0.001 0.005 0.005"/>
    </default>
  <worldbody>
 <body name="base" pos="0 0 -0.5">
      <geom type="box" size="0.3 0.3 0.01" rgba="0.5 0.5 0.5 1"/>
    </body>
    <body name="pulley" pos="0 0 0">
      <geom name="pulley" type="cylinder" pos="0 0 0.5" quat="0.74 0.74 0 0"  solimp="0.9 0.95 0.001" solref="0.0001 1" size="0.05 0.02"  rgba="0.8 0.8 0.8 0.2" />
      <site name="site1" type="sphere" size="0.005" pos="0.05 0 0.5" rgba="1 0 0 1"/>
       <site name="site2" type="sphere" size="0.005" pos="-0.05 0 0.5" rgba="0 1 0 1"/>
      <site name="pulley1"  size="0.005" pos="0 0 .6"/>  
    </body>
    <body name="left_rope" pos="-0.2 0 0.4">
      <site name="left_rope" size="0.02"   pos="0 0 0" rgba="1 0 0 1" />
    </body>
    <body name="right_weight" pos="0.2 0 0.6">
      <joint type="free" axis="0 0 1" />
     <geom type="box" size="0.05 0.05 0.05"    mass="0.01" rgba="0 0 1 1"/>
       <site name="box" size="0.01"  pos="0 0 .05" rgba="1 0 0 1" />
    </body>
  </worldbody>
  <tendon>
    <spatial name="rope1" limited="true" range="0 0.7" solimplimit="0.001 0.001 0.001 0.5 1" solreflimit="-100 -0">
      <site site="left_rope"/>
      <geom geom="pulley" sidesite="pulley1"/>
      <site site="box"/>
    </spatial>
  </tendon>
</mujoco>

### Code required for reproduction

```python
import mujoco
import mediapy as media
model = mujoco.MjModel.from_xml_string(xml)
data = mujoco.MjData(model)

with mujoco.Renderer(model) as renderer:
  mujoco.mj_forward(model, data)
  renderer.update_scene(data)

  media.show_image(renderer.render())
```

### Confirmations

- [x] I searched the [latest documentation](https://mujoco.readthedocs.io/en/latest/overview.html) thoroughly before posting.
- [x] I searched previous [Issues](https://github.com/google-deepmind/mujoco/issues) and [Discussions](https://github.com/google-deepmind/mujoco/discussions), I am certain this has not been raised before.

tttj12, 2025-04-16:
<mujoco>
  <default >
    <joint armature="1e-05" damping="0.05"/>
    <site size="0.001 0.005 0.005"/>
    </default>
  <worldbody>
 <body name="base" pos="0 0 -0.5">
      <geom type="box" size="0.3 0.3 0.01" rgba="0.5 0.5 0.5 1"/>
    </body>
    <body name="pulley" pos="0 0 0">
      <geom name="pulley" type="cylinder" pos="0 0 0.5" quat="0.74 0.74 0 0"  solimp="0.9 0.95 0.001" solref="0.0001 1" size="0.05 0.02"  rgba="0.8 0.8 0.8 0.2" />
      <site name="site1" type="sphere" size="0.005" pos="0.05 0 0.5" rgba="1 0 0 1"/>
       <site name="site2" type="sphere" size="0.005" pos="-0.05 0 0.5" rgba="0 1 0 1"/>
      <site name="pulley1"  size="0.005" pos="0 0 .6"/>  
    </body>
    <body name="left_rope" pos="-0.2 0 0.4">
      <site name="left_rope" size="0.02"   pos="0 0 0" rgba="1 0 0 1" />
    </body>
    <body name="right_weight" pos="0.2 0 0.6">
      <joint type="free" axis="0 0 1" />
     <geom type="box" size="0.05 0.05 0.05"    mass="0.01" rgba="0 0 1 1"/>
       <site name="box" size="0.01"  pos="0 0 .05" rgba="1 0 0 1" />
    </body>
  </worldbody>
  <tendon>
    <spatial name="rope1" limited="true" range="0 0.7" solimplimit="0.001 0.001 0.001 0.5 1" solreflimit="-100 -0">
      <site site="left_rope"/>
      <geom geom="pulley" sidesite="pulley1"/>
      <site site="box"/>
    </spatial>
  </tendon>
</mujoco>

tttj12, 2025-04-16:
[1.xml.txt](https://github.com/user-attachments/files/19772066/1.xml.txt)

tttj12, 2025-04-16:
Sorry. Later, we recalculated the length of the tendon, and it is in line with the length of the arc that includes the part around the cylinder. The tendon passing through the cylinder is a visual effect. Since we are beginners, we would like to ask you something. Is it feasible to use the tendon to drive the manipulator and simulate the elongation of the tendon to be the same as the elongation of the rope driven by the motor in reality? Do we need to make any changes to the tendon statements?

Balint-H, 2025-05-08:
Hello! Tendon actuated robotics are definitely a feasible with MuJoCo. You may wish to consider settings for the `range`, `solimplimit`, and `solreflimit` attributes in your tendons (see https://mujoco.readthedocs.io/en/stable/XMLreference.html#tendon-spatial-solimplimit). A lot of aspects of tendon actuation  (gains, stiffness, control limits) will be configured in how you set up your `actuator` elements (see https://mujoco.readthedocs.io/en/stable/XMLreference.html#actuator). 

Let me know if you need more info to get started!