google-deepmind/mujoco #2585: Compiling Mujoco to wasm with emscripten

LazyYuuki, 2025-04-19:
### The feature, motivation and pitch

I'm working on compiling Mujoco to wasm using emscripten and would like to be able to make the Mujoco rencer to match with GLES2 instead of GL.
 
I have consulted:

https://github.com/zalo/mujoco_wasm

For direction on how I can do so, but the method proposed by Zalo above is not efficient and native. It requires the mapping of Mujoco to Three.js function to then be able to render in the browser, while theoretically, you should be able to just run OpenGL map to WebGL directly through GLES2, reducing 1 layer of abstraction that would require more maintenance otherwise.

Furthermore, zalo guide is under the assumption that you already successfully compile Mujoco to static archive to be able to compile to wasm, however, the guide and version they based on is now outdated, and I have provided a new method as detailed down here (albeit incomplete and might contain mistake as I am a novice to all of this):

https://github.com/menloresearch/mujoco-wasm

The main problem I am having right now is that the current way Mujoco handling OpenGL rendering is by using `glad` to generate a loader for OpenGL v 1.5. I can built the source code, but when I invoke the binary to call for `mjr_updateContext` it will throw an error back saying `cannot find dl_pthread_iterate`. Upon further digging, I realise that it is because it is dynamically trying to find this function in a shared library on Linux system instead, and could not find it.

Further research show me that, to make this compatible easier, it is better for me to go and generate a new `glad` loader specifically for GLES2  using this folllowing link:

https://gen.glad.sh/

Here is my setting after generated the file:

 * Loader generated by glad 2.0.8 on Sat Apr 19 02:15:03 2025
 *
 * SPDX-License-Identifier: (WTFPL OR CC0-1.0) AND Apache-2.0
 *
 * Generator: C/C++
 * Specification: gl
 * Extensions: 45
 *
 * APIs:
 *  - gles2=3.2
 *
 * Options:
 *  - ALIAS = True
 *  - DEBUG = False
 *  - HEADER_ONLY = False
 *  - LOADER = True
 *  - MX = True
 *  - ON_DEMAND = False

As far as I know, I don't include any extensions for now for stability reason.

Right now I am stuck at the part where I can see that there is quite a bit of modifciation to change all the `gl` function call to become `mjGl_` instead, and it is causing confusion for me to know how should I approach this modification.

I would really appreciate if I can get some help or guidance on how should I proceed on what I need to change and where. Once I get it workings, the code will be subsequently free and open for everyone else to use also on the github link provided above. It would be very exciting to make Mujoco even more portable and available to more platform to be used on.

LazyYuuki, 2025-04-19:
https://glad.dav1d.de/

I just discovered this after writing the above and it seems to be much closer to what we have being used in MuJoCo after I choose the same option for the built for GLES2.

I will work further on this and give update as I make progress.

yunho-c, 2025-08-22:
How is it coming along? I am very excited about the possibility of having MuJoCo @ WASM and was just curious :D

yuvaltassa, 2025-08-22:
It's going very well, we're nearly done. 
The biggest challenge now is the packaging. 
I expect this to be released in October.

yunho-c, 2025-08-28:
That's fantastic, thank you for the response!!!

okmatija, 2025-10-19:
I think we'll be able to make an initial release this week :)

okmatija, 2025-10-28:
It's almost done! There is just a bit more infrastructure to finish... we have to set up transformations between the github source code and the internal google source code, also, there are some CI scripts which aren't quite working yet. I'm hoping it'll be done today or tomorrow.

closed 2025-10-31 by yuvaltassa

zalo, 2025-11-01:
For those too lazy to build it, [here is the built wasm](https://github.com/zalo/mujoco/actions/runs/18987848251/artifacts/4435233160) and here's a video of the current demo:

https://github.com/user-attachments/assets/a00efc1e-396b-40e2-a5f1-6813dffed1b8

Great start!

LazyYuuki, 2026-02-24:
I know I am a bit late to the party now (almost 1 year late), but will there be worked done toward changing from GL to GLES2 binding so that we can directly use WebGL in the browser instead of having to bind it through Three.js or like a thirdparty renderer?

Thanks 🙏🏼 

cc @yuvaltassa