google-deepmind/mujoco #2606: Having a mocap body in scene changes stability of actuation

Balint-H, 2025-04-30:
### Intro

Hi!

I am using MuJoCo to study human motor control. 

### My setup

Using MJC primarily on Windows and MJX primarily on Linux, both with the python API. Version 3.3.2.

### What's happening? What did you expect?

Adding a disjoint body in an MJX scene with `mocap=True` causes the actuation of unrelated kinematic trees to become unstable. The same scene either in MJC or with `mocap=False` acts as expected. This was observed primarily for muscle actuated models.

### Steps for reproduction

Added a MRE script below that uses the passive viewer of MJC to visualize the effect. If MJX is used and a mocap body is present, the actuation becomes noisy with a high frequency. The same issue is not observed however with a non-tendon actuated model.

### Code required for reproduction

<details>
<summary>MRE</summary>

``` python
import time
import mujoco
from etils import epath
from mujoco import mjx
import mujoco.viewer
import jax

USE_HUMANOID = False

path = (epath.Path(epath.resource_path('mujoco')) / (
        'mjx/test_data/humanoid/01_humanoids.xml')).as_posix() \
  if USE_HUMANOID else (epath.Path(epath.resource_path('mujoco')) / (
        'mjx/test_data/actuator/arm26.xml')).as_posix()

USE_MJX = True
MOCAP_IN_SCENE = True

def main():
    spec = mujoco.MjSpec.from_file(path)

    spec.option.iterations = 20
    spec.option.ls_iterations = 10

    mocap = spec.worldbody.add_body(name="mocap_body", mocap=MOCAP_IN_SCENE, pos=[0, 1.5, 0])
    mocap.add_geom(name="mocap_geom", type=mujoco.mjtGeom.mjGEOM_SPHERE, size=[0.1, 0, 0], rgba=[0.5, 0.1, 0.1, 0.1])

    # We'll visualize in MJC
    m = spec.compile()
    d = mujoco.MjData(m)

    # Physics in MJX
    mjx_m = mjx.put_model(m)
    mjx_d = mjx.make_data(m)

    jit_step = jax.jit(mjx.step)
    rng, key = jax.random.split(jax.random.key(0), 2)

    # We'll change ctrl only once every 1000 steps
    downsample = 1000
    countdown = downsample

    ctrl = jax.random.uniform(rng, m.nu)
    with mujoco.viewer.launch_passive(m, d) as viewer:
        while viewer.is_running():
            step_start = time.time()
            rng, key = jax.random.split(key, 2)

            if countdown < 1:
              countdown = downsample
              ctrl = jax.random.uniform(rng, m.nu)*2-1 if USE_HUMANOID else jax.random.uniform(rng, m.nu)
            else:
              countdown -= 1

            if USE_MJX:
              mjx_d = mjx_d.replace(ctrl=ctrl)
              mjx_d = jit_step(mjx_m, mjx_d)

              # Render pose + display ctrl on sidebar
              d.qpos = mjx_d.qpos
              mujoco.mj_fwdPosition(m, d)
              d.ctrl[:] = ctrl[:]
            else:
              d.ctrl[:] = ctrl[:]
              mujoco.mj_step(m, d)

            viewer.sync()

            # Rudimentary time keeping, will drift relative to wall clock.
            time_until_next_step = m.opt.timestep - (time.time() - step_start)
            if time_until_next_step > 0:
                time.sleep(time_until_next_step)
    pass

if __name__ == '__main__':
    main()
```

</details>

### Confirmations

- [x] I searched the [latest documentation](https://mujoco.readthedocs.io/en/latest/overview.html) thoroughly before posting.
- [x] I searched previous [Issues](https://github.com/google-deepmind/mujoco/issues) and [Discussions](https://github.com/google-deepmind/mujoco/discussions), I am certain this has not been raised before.

Balint-H, 2025-05-01:
@kevinzakka suggested something interesting: If collisions are disabled for the geom on the mocap body (via `contype` and `conaffinity`) the high frequency noise stops again.

yuvaltassa, 2025-05-01:
Are you sure this is not in the "surprising collisions" category?

For example, a minimal example that shows that collisons exist in MJC but not in MJX, would be really helpful. In particular, and example that is defined in pure, short XML, no includes/attaches from files. Like, actually minimal 🙂

Balint-H, 2025-05-01:
Thanks for the reply!

Directly comparing the contacts is a good idea. The bug seems to be MJX not using the mocap_pos field correctly. `make_data` zeros the field and doesn't set it from the model, but even when created with `put_data` there are collisions.

This is not a static parent - type collision (if I understand correctly what you are referring to) as the collisions are between parts of the kinematic chain and a separate body.

Sorry for not embedding the XML string but rather using the built-in models in the python module, I did not know that was advised against for issues. Also, initially I had no idea this was related to collisions, which is why I sent a setup with a simulate window where the unexpected phenomenon could be observed and reproduced.

In any case here's a MRE demonstrating the new insights:

``` python
import mujoco
from mujoco import mjx
import mujoco.viewer
import jax

xml = (
"""
<mujoco model="MRE_mocap_collisions">

  <worldbody>
    <body pos="0 0 0">
      <geom name="dynamic_geom" type="capsule" size="0.045" fromto="0 0 0  0.5 0 0"/>
      <joint/>
    </body>
    <body mocap="true" pos="1 1 0">
      <geom size="0.2" type="sphere"/>
    </body>
  </worldbody>

</mujoco>
"""
)

def main():
    m = mujoco.MjModel.from_xml_string(xml)
    d = mujoco.MjData(m)

    mjx_m = mjx.put_model(m)
    mjx_d = mjx.make_data(m, d)

    print(d.mocap_pos)
    print(mjx_d.mocap_pos)
    print("\n")

    jit_step = jax.jit(mjx.step)

    mujoco.mj_step(m, d)
    mjx_d = jit_step(mjx_m, mjx_d)

    print(mjx_d.contact)
    print("\n")
    print(d.contact)
    print("\n")
    print(d.xpos[-1])
    print(mjx_d.xpos[-1])
    print("\n")
    print(d.mocap_pos)
    print(mjx_d.mocap_pos)
    pass

if __name__ == '__main__':
    main()

```

btaba, 2025-05-06:
Thanks @Balint-H for the minimal repro. Fix is on the way!

commit 421c487d (2025-05-07): Set mocap_pos/quat in make_data. Fixes #2606.

PiperOrigin-RevId: 755903897
Change-Id: Ied04e9bd3febfe1bbbba24ed1f52b3c0c6e50f38