google-deepmind/mujoco #2610: Simulation error using mjx

Hang-Hu1, 2025-05-06:
### Intro

Hi!

I am a student and I'm using Mujoco (especially MJX) for rigid body simulation and reinforcement learning. 

### My setup

I used both version 3.3.0 and the remote environment in the tutorial notebook in this link [https://mujoco.readthedocs.io/en/stable/mjx.html](https://mujoco.readthedocs.io/en/stable/mjx.html).

### What's happening? What did you expect?

I'm encountering GPU simulation distortion issues very frequently. When objects make contact with the ground, they either bounce abnormally or immediately result in NaN values in data.xpos and qpos. In any case, the results are significantly different from those of the CPU simulation. Reducing the time step or changing the integrator does not solve the problem.

Even a very simple example (even a simple capsule falling to the ground causes such anomalies) is able to produce this error.

### Steps for reproduction

Just need to run the following code.

### Minimal model for reproduction

```XML
xml = """
<?xml version="1.0" ?>
<mujoco>
    <default>
        <site rgba=".9 .9 .9 1"/>
    </default>
    <asset>
        <texture name="grid" type="2d" builtin="checker" rgb1=".1 .2 .3" rgb2=".2 .3 .4" width="300" height="300" mark="none"/>
        <material name="grid" texture="grid" texrepeat="6 6" texuniform="true" reflectance=".2"/>
    </asset>
    <option timestep="0.0001">
        <flag energy="disable" contact="enable" gravity="enable"/>
    </option>
    <worldbody>
        <light name="light" pos="-.2 0 1"/>
        <geom name="ground" type="plane" pos="0 0 -0.001" size="0 0 10" material="grid" zaxis="0 0 1" friction="0.0"/>
        <camera name="cam" pos="0 0 7"/>
        <light pos="0 0 1"/>
        <light pos="-1 0 1"/>
        <light pos="-2 0 1"/>
        <light pos="-3 0 1"/>
        <light pos="-4 0 1"/>
        <light pos="-5 0 1"/>
        <body name="b1" pos="0 0 0.05">
            <joint name="j1" type="free"/>
            <geom friction="0" name="g1" type="capsule" size="0.04" fromto="0 0 0 0.19855637987881708 0 0"/>
        </body>
    </worldbody>
    <tendon/>
</mujoco>

"""

### Code required for reproduction

```Python
import time
import numpy as np
import jax
import jax.numpy as jp
import matplotlib.pyplot as plt
import mujoco
from mujoco import mjx

mj_model = mujoco.MjModel.from_xml_string(xml)
mj_data = mujoco.MjData(mj_model)

duration = 0.5  # (seconds)
framerate = 24  # (Hz)

jit_step = jax.jit(mjx.step)

frames = []
mujoco.mj_resetData(mj_model, mj_data)
mjx_model = mjx.put_model(mj_model)
mjx_data = mjx.put_data(mj_model, mj_data)

while mjx_data.time < duration:
    # Step.
    mjx_data = jit_step(mjx_model, mjx_data)
    print("xpos", jp.sum(mjx_data.xpos))

### Confirmations

- [x] I searched the [latest documentation](https://mujoco.readthedocs.io/en/latest/overview.html) thoroughly before posting.
- [x] I searched previous [Issues](https://github.com/google-deepmind/mujoco/issues) and [Discussions](https://github.com/google-deepmind/mujoco/discussions), I am certain this has not been raised before.

Hang-Hu1, 2025-05-06:
I've noticed that there's randomness when reproducing this bug; sometimes the simulation doesn't fail. If that's the case, the following XML can be considered.

```XML
<?xml version="1.0" ?>
<mujoco>
    <default>
        <site rgba=".9 .9 .9 1"/>
    </default>
    <asset>
        <texture name="grid" type="2d" builtin="checker" rgb1=".1 .2 .3" rgb2=".2 .3 .4" width="300" height="300" mark="none"/>
        <material name="grid" texture="grid" texrepeat="6 6" texuniform="true" reflectance=".2"/>
    </asset>
    <option timestep="0.0001">
        <flag energy="disable" contact="enable" gravity="enable"/>
    </option>
    <worldbody>
        <light name="light" pos="-.2 0 1"/>
        <geom name="ground" type="plane" pos="0 0 0" size="0 0 10" material="grid" zaxis="0 0 1" friction="0.0"/>
        <camera name="cam" pos="0 0 7"/>
        <light pos="0 0 1"/>
        <light pos="-1 0 1"/>
        <light pos="-2 0 1"/>
        <light pos="-3 0 1"/>
        <light pos="-4 0 1"/>
        <light pos="-5 0 1"/>
        <body name="b1" pos="0 0 0.05">
            <joint name="j1" type="free"/>
            <geom friction="0" name="g1" type="capsule" size="0.05" fromto="0 0 0 0.19855637987881708 0 0"/>
            <body name="b900001" pos="0.2551249223737409 0.05656854249492381 0">
                <joint name="j900001" type="hinge" axis="0 0 1" pos="-0.028284271247461905 -0.028284271247461905 0" range="-20 20"/>
                <geom friction="0" name="g900001" type="capsule" size="0.05" fromto="0 0 0 0.09685116991281727 0.09685116991281727 0"/>
            </body>
        </body>
    </worldbody>
    <tendon/>
    <actuator>
        <motor name="motor10" gear="4" joint="j900001"/>
    </actuator>
</mujoco>

btaba, 2025-05-06:
Hi @Hang-Hu1 , it looks like friction=0 is causing NaNs. If you set values like 0.01, the issue goes away (note I also added these options `iterations="3" ls_iterations="5" timestep="0.001"` and `eulerdamp="disable"`). CC @erikfrey @thowell if you'd like to take a look to make the solver more numerically robust to settings like these.

Hang-Hu1, 2025-05-07:
This solution resolved the issue I was facing, and I truly thank you for your response!

closed 2025-05-28 by btaba