google-deepmind/mujoco #2618: [C API] Option for static linking, for better performance

davidhozic, 2025-05-11:
### The feature, motivation and pitch

I would like to propose a flag inside CMakeLists.txt that would allow us to statically link MuJoCo (i. e., produce a libmujoco.a). It doesn't sound like this would be beneficial at first, but upon testing I'm getting roughtly **+34 % steps more steps per second** compared to the dynamically linked library (~57000 -> ~76500).

### Additional context

The performence gains were measured on the Ryzen 5 5600X CPU and in a thin Rust wrapper around the MuJoCo's C API.

saran-t, 2025-06-17:
Do you want to send us a PR so that we can see what this would look like? No objection in principle, we also link statically for internal users.

saran-t, 2025-06-17:
Although the 34% performance increase is _very_ strange, that shouldn't be a thing at all. Have you looked deeper into what's causing this? Are you comparing your custom static lib build against our prebuilt, or did you use your own custom DSO build too? Can you run the benchmark using our canonical binary as a baseline too?

davidhozic, 2025-06-18:
I'll prepare a PR later today and do some extra benchmarks.