google-deepmind/mujoco #2630: Inconsistent mj_contactforce calculation with theoretical results

Chuanfang-Neptune, 2025-05-14:
### Intro

Hi!

I am a Phd student using MuJoCo to simulate some quadrupod robots. I want to reduce the reality gap between MuJoCo simulation and my real robot experiments, and I'm validating some metrics like the contact force to match with real sensor readings.

I try to match the contact force from `mujoco.mj_contactForce` with my real robot and the value was 1-2 magnitudes smaller. Therefore I tried to compute the force mathematically from [solver parameters](https://mujoco.readthedocs.io/en/latest/modeling.html#solver-parameters) to see what is going wrong. 

In this relevant issue [#732](https://github.com/google-deepmind/mujoco/issues/732) the OP tried to validate the penetration at steady state and compared with real results (which was matching), while I want to match the force before reaching steady state.

From the documentation, MuJoCo didn't explicitly compute contact force but computed a target acceleration with $a_1+d⋅(bv+kr)=(1−d)⋅a_0 $. Meanwhile MuJoCo provides some function such as 'mj_contactForce' that put the data in `physics.sensordata` that can be accessed later. I did a simulation experiment to test if mj_contactForce is computed as in the documentation.

### My setup

mujoco: 3.1.6
tested both on Windows and Ubuntu

### What's happening? What did you expect?

Here I use `solref = "0.02 1"` as default value, equivalent to `K = 1/0.02**2` and `D = 2*1/0.02` in direct format. I use `solimp = 0.9 0.95 0.001 0.5 2`, where d will be between 0.9 and 0.95. 

If $a_0$ is acceleration in absence of the contact, it should be `-9.8m/s^2` from gravity. And the $a_1$ should be the acceleration with contact. Therefore, I suppose, the contact force should be $m*(a_1-a_0)$, which caused the change of acceleration (here only considering normal force (z)). Embedding the contact force in the equation above yields $F_{contact} =ma_1-ma_0= -md(g+bv+kr)$

I wrote a script where I set a cube to start at a position below the ground with different mass, position and velocity. I do one `mj_step` to update the physics and record the contact force (`mj_contactForce`) to compare with the theoretical value and print the error.

I expect the error to be small but the error is several times larger than the expected force.
```
Mass: 1.0 kg, Exp: Depth: 0.0020, Speed: 0.00 → expforce: 3.72 theoryforce: 14.06 error:73.55%
Mass: 1.0 kg, Exp: Depth: 0.0020, Speed: -0.50 → expforce: 16.71 theoryforce: 61.56 error:72.86%
Mass: 1.0 kg, Exp: Depth: 0.0020, Speed: -1.00 → expforce: 29.69 theoryforce: 109.06 error:72.77%
Mass: 1.0 kg, Exp: Depth: 0.0040, Speed: 0.00 → expforce: 5.02 theoryforce: 18.81 error:73.32%
Mass: 1.0 kg, Exp: Depth: 0.0040, Speed: -0.50 → expforce: 18.01 theoryforce: 66.31 error:72.85%
Mass: 1.0 kg, Exp: Depth: 0.0040, Speed: -1.00 → expforce: 30.99 theoryforce: 113.81 error:72.77%
Mass: 1.0 kg, Exp: Depth: 0.0060, Speed: 0.00 → expforce: 6.32 theoryforce: 23.56 error:73.19%
Mass: 1.0 kg, Exp: Depth: 0.0060, Speed: -0.50 → expforce: 19.30 theoryforce: 71.06 error:72.83%
Mass: 1.0 kg, Exp: Depth: 0.0060, Speed: -1.00 → expforce: 32.29 theoryforce: 118.56 error:72.76%
Mass: 1.0 kg, Exp: Depth: 0.0080, Speed: 0.00 → expforce: 7.62 theoryforce: 28.31 error:73.10%
Mass: 1.0 kg, Exp: Depth: 0.0080, Speed: -0.50 → expforce: 20.60 theoryforce: 75.81 error:72.82%
Mass: 1.0 kg, Exp: Depth: 0.0080, Speed: -1.00 → expforce: 33.59 theoryforce: 123.31 error:72.76%
Mass: 1.0 kg, Exp: Depth: 0.0100, Speed: 0.00 → expforce: 8.91 theoryforce: 33.06 error:73.04%
Mass: 1.0 kg, Exp: Depth: 0.0100, Speed: -0.50 → expforce: 21.90 theoryforce: 80.56 error:72.81%
Mass: 1.0 kg, Exp: Depth: 0.0100, Speed: -1.00 → expforce: 34.89 theoryforce: 128.06 error:72.76%
Mass: 2.0 kg, Exp: Depth: 0.0020, Speed: 0.00 → expforce: 6.14 theoryforce: 28.12 error:78.17%
Mass: 2.0 kg, Exp: Depth: 0.0020, Speed: -0.50 → expforce: 19.13 theoryforce: 123.12 error:84.46%
Mass: 2.0 kg, Exp: Depth: 0.0020, Speed: -1.00 → expforce: 32.11 theoryforce: 218.12 error:85.28%
Mass: 2.0 kg, Exp: Depth: 0.0040, Speed: 0.00 → expforce: 7.44 theoryforce: 37.62 error:80.23%
Mass: 2.0 kg, Exp: Depth: 0.0040, Speed: -0.50 → expforce: 20.43 theoryforce: 132.62 error:84.60%
Mass: 2.0 kg, Exp: Depth: 0.0040, Speed: -1.00 → expforce: 33.41 theoryforce: 227.62 error:85.32%
Mass: 2.0 kg, Exp: Depth: 0.0060, Speed: 0.00 → expforce: 8.74 theoryforce: 47.12 error:81.46%
Mass: 2.0 kg, Exp: Depth: 0.0060, Speed: -0.50 → expforce: 21.72 theoryforce: 142.12 error:84.71%
Mass: 2.0 kg, Exp: Depth: 0.0060, Speed: -1.00 → expforce: 34.71 theoryforce: 237.12 error:85.36%
Mass: 2.0 kg, Exp: Depth: 0.0080, Speed: 0.00 → expforce: 10.04 theoryforce: 56.62 error:82.27%
Mass: 2.0 kg, Exp: Depth: 0.0080, Speed: -0.50 → expforce: 23.02 theoryforce: 151.62 error:84.82%
Mass: 2.0 kg, Exp: Depth: 0.0080, Speed: -1.00 → expforce: 36.01 theoryforce: 246.62 error:85.40%
Mass: 2.0 kg, Exp: Depth: 0.0100, Speed: 0.00 → expforce: 11.33 theoryforce: 66.12 error:82.86%
Mass: 2.0 kg, Exp: Depth: 0.0100, Speed: -0.50 → expforce: 24.32 theoryforce: 161.12 error:84.90%
Mass: 2.0 kg, Exp: Depth: 0.0100, Speed: -1.00 → expforce: 37.31 theoryforce: 256.12 error:85.43%
Mass: 3.0 kg, Exp: Depth: 0.0020, Speed: 0.00 → expforce: 8.56 theoryforce: 42.18 error:79.70%
Mass: 3.0 kg, Exp: Depth: 0.0020, Speed: -0.50 → expforce: 21.55 theoryforce: 184.68 error:88.33%
Mass: 3.0 kg, Exp: Depth: 0.0020, Speed: -1.00 → expforce: 34.53 theoryforce: 327.18 error:89.44%
Mass: 3.0 kg, Exp: Depth: 0.0040, Speed: 0.00 → expforce: 9.86 theoryforce: 56.43 error:82.53%
Mass: 3.0 kg, Exp: Depth: 0.0040, Speed: -0.50 → expforce: 22.85 theoryforce: 198.93 error:88.52%
Mass: 3.0 kg, Exp: Depth: 0.0040, Speed: -1.00 → expforce: 35.83 theoryforce: 341.43 error:89.50%
Mass: 3.0 kg, Exp: Depth: 0.0060, Speed: 0.00 → expforce: 11.16 theoryforce: 70.68 error:84.21%
Mass: 3.0 kg, Exp: Depth: 0.0060, Speed: -0.50 → expforce: 24.15 theoryforce: 213.18 error:88.67%
Mass: 3.0 kg, Exp: Depth: 0.0060, Speed: -1.00 → expforce: 37.13 theoryforce: 355.68 error:89.56%
Mass: 3.0 kg, Exp: Depth: 0.0080, Speed: 0.00 → expforce: 12.46 theoryforce: 84.93 error:85.33%
Mass: 3.0 kg, Exp: Depth: 0.0080, Speed: -0.50 → expforce: 25.44 theoryforce: 227.43 error:88.81%
Mass: 3.0 kg, Exp: Depth: 0.0080, Speed: -1.00 → expforce: 38.43 theoryforce: 369.93 error:89.61%
Mass: 3.0 kg, Exp: Depth: 0.0100, Speed: 0.00 → expforce: 13.76 theoryforce: 99.18 error:86.13%
[... 56 lines cut ...]
            # Get contact force (notmal z)
            force_val = 0
            if data.ncon > 0:
                contact = data.contact[0]
                force = np.zeros(6)
                mujoco.mj_contactForce(model, data, 0, force)
                force_val = force[0]  # normal force

            forces[i, j] = force_val
            
            
            # Theoretical force
            # convert kd to indirect format#2629 
            # d=0.95~0.9. take 0.95here
            K = 1/0.02**2 
            D = 2*1/0.02
            expected_force = 0.95*m* (9.8 + K * depth + D * abs(speed))

            print(f"Mass: {m:.1f} kg, Exp: Depth: {depth:.4f}, Speed: {speed:.2f} → expforce: {force_val:.2f} theoryforce: {expected_force:.2f} error:{abs(expected_force-force_val)/expected_force*100:.2f}%")
```

### Confirmations

- [x] I searched the [latest documentation](https://mujoco.readthedocs.io/en/latest/overview.html) thoroughly before posting.
- [x] I searched previous [Issues](https://github.com/google-deepmind/mujoco/issues) and [Discussions](https://github.com/google-deepmind/mujoco/discussions), I am certain this has not been raised before.

sunilgora, 2025-05-16:
Hi @Chuanfang-Neptune , I would suggest to use the following in your code:

1. $k=\frac{d(r)}{d_\text{width}^2 \cdot \text{timeconst}^2 \cdot \text{dampratio}^2}$
2. $b=\frac{2}{d_\text{width} \cdot \text{timeconst}}$
3. $a_{ref}=k\cdot \text{depth}- d\cdot \text{speed}$
4. $A=JM^{-1}J^T$
5. $R=\frac{1-d}{d}A$
6. $f=(A+R)^{-1}(a_{ref}-a_0)$This should give the correct theoretical force.

You can take $A=$`data.efc_diagApprox[0]` and $a_0=-9.81$ in your code, which will give the following results:

```
Mass: 1.0 kg, Exp: Depth: 0.0020, Speed: 0.00 → expforce: 3.72 theoryforce: 3.58 error:3.90%
Mass: 1.0 kg, Exp: Depth: 0.0020, Speed: -0.50 → expforce: 16.71 theoryforce: 16.08 error:3.90%
Mass: 1.0 kg, Exp: Depth: 0.0020, Speed: -1.00 → expforce: 29.69 theoryforce: 28.58 error:3.90%
Mass: 1.0 kg, Exp: Depth: 0.0040, Speed: 0.00 → expforce: 5.02 theoryforce: 4.83 error:3.90%
Mass: 1.0 kg, Exp: Depth: 0.0040, Speed: -0.50 → expforce: 18.01 theoryforce: 17.33 error:3.90%
Mass: 1.0 kg, Exp: Depth: 0.0040, Speed: -1.00 → expforce: 30.99 theoryforce: 29.83 error:3.90%
Mass: 1.0 kg, Exp: Depth: 0.0060, Speed: 0.00 → expforce: 6.32 theoryforce: 6.08 error:3.90%
Mass: 1.0 kg, Exp: Depth: 0.0060, Speed: -0.50 → expforce: 19.30 theoryforce: 18.58 error:3.90%
Mass: 1.0 kg, Exp: Depth: 0.0060, Speed: -1.00 → expforce: 32.29 theoryforce: 31.08 error:3.90%
Mass: 1.0 kg, Exp: Depth: 0.0080, Speed: 0.00 → expforce: 7.62 theoryforce: 7.33 error:3.90%
Mass: 1.0 kg, Exp: Depth: 0.0080, Speed: -0.50 → expforce: 20.60 theoryforce: 19.83 error:3.90%
Mass: 1.0 kg, Exp: Depth: 0.0080, Speed: -1.00 → expforce: 33.59 theoryforce: 32.33 error:3.90%
Mass: 1.0 kg, Exp: Depth: 0.0100, Speed: 0.00 → expforce: 8.91 theoryforce: 8.58 error:3.90%
Mass: 1.0 kg, Exp: Depth: 0.0100, Speed: -0.50 → expforce: 21.90 theoryforce: 21.08 error:3.90%
Mass: 1.0 kg, Exp: Depth: 0.0100, Speed: -1.00 → expforce: 34.89 theoryforce: 33.58 error:3.90%
```
Also, you can not just change the value of `model.body_mass[body_id] = m` to change the mass, you need to change it in the xml and load the xml model again.
Run this code to get the correct results:
```
import mujoco
import numpy as np

# create a cube with 0.05m side
# SET solref to 0.02 1
# SET solimp to 0.99 0.95 0.001 0.5 2

mjcf = """
<mujoco>
    <option gravity="0 0 -9.81"/>
    <worldbody>
        <geom name="floor" type="plane" pos="0 0 0" mass="1" size="1 1 0.1" rgba="0.8 0.9 0.8 1"/>
        <body name="cube" pos="0 0 0.2">
            <joint name="free" type="free"/>
            <geom name="box" type="box" mass="1" size="0.05 0.05 0.05" rgba="1 0 0 1"
                  solref="0.02 1" solimp="0.99 0.95 0.001 0.5 2" contype="1" conaffinity="1"/>
        </body>
    </worldbody>
</mujoco>
"""

model = mujoco.MjModel.from_xml_string(mjcf)
data = mujoco.MjData(model)

body_id = mujoco.mj_name2id(model, mujoco.mjtObj.mjOBJ_BODY, "cube")

# Grid of penetration depths and speeds
depths = np.linspace(0.002, 0.01, 5)  # meters
speeds = np.linspace(0.0, -1.0, 3)  # m/s (negative: downward)
mass = np.linspace(1.0, 1.0, 1)
forces = np.zeros((len(depths), len(speeds)))
for m in mass:
    model.body_mass[body_id] = m
    for i, depth in enumerate(depths):
        for j, speed in enumerate(speeds):
            # Reset sim
            data = mujoco.MjData(model)

            # teleport cube below surface (z = 0.05 box half-height - depth)
            cube_z = 0.05 - depth
            data.qpos[2] = cube_z
            data.qvel[2] = speed

            # one step to update force
            mujoco.mj_forward(model, data)
            # Get contact force (notmal z)
            force_val = 0
            if data.ncon > 0:
                contact = data.contact[0]
                force = np.zeros(6)
                mujoco.mj_contactForce(model, data, 0, force)
                force_val = force[0]  # normal force

            forces[i, j] = force_val

            # Theoretical force
            A=data.efc_diagApprox[0]
            R=(1-0.95)/0.95*A
            # convert kd to indirect format#2629
            # d=0.95~0.9. take 0.95here
            K = 1/(0.02**2)/0.95
            D = 2*1/0.02/0.95
            expected_force = 1/(A+R) * ( 1*( K * depth - D * (speed)) -(- 9.81) )

            print(f"Mass: {m:.1f} kg, Exp: Depth: {depth:.4f}, Speed: {speed:.2f} → expforce: {force_val:.2f} theoryforce: {expected_force:.2f} error:{abs(expected_force-force_val)/expected_force*100:.2f}%")
```

yuvaltassa, 2025-05-16:
Sounds like @sunilgora has resolved the question? (thanks, @sunilgora !!)

Chuanfang-Neptune, 2025-05-16:
Thank you very much @sunilgora! It matches!