google-deepmind/mujoco #2659: AVX-512 causes a non-reproducibility in mujoco

TimotheeMathieu, 2025-05-29:
### Intro

Hi!

I am a researcher at Inria (France), I use MuJoCo for my research on RL.

### My setup

Mujoco Version 3.3.1, python API, on Ubuntu x86_64 

### What's happening? What did you expect?

When I use mujoco on a computer with AVX-512 I get a result that is different result as on a computer without AVX-512. 

- To be sure that it was due to AVX-512, I used a vm with qemu and enabling/disabling AVX-512 gave a different result with the exact same script environment and setup (other than AVX-512 enabled in one and disabled in the other). 
- I checked that otherwise my script was reproducible giving exactly the same result between two computers with avx-512. It is also completely reproducible between two non-avx-512 computers.
- I took care to use a fixed seed, and to use `th.use_deterministic_algorithms(True)` for rl algo reproducibility (I use stablebaselines3).
- I don't think that sb3 algos were the problem as the non-reproducibility is not present on Pendulum from gymnasium or other classic control environments for which I have 100% reproducible results (i.e. exactly same float representation of final reward).
- The results are very different with up to 5 reward difference between two evaluations episodes of 1000 steps (on Ant). 
- On simple environments such as Reacher, Inverted pendulum, inverted double pendulum the results are different but not by much so I guess the contact are what makes a big difference.
- I tried to compile mujoco without avx (with  `-DMUJOCO_ENABLE_AVX_INTRINSICS=OFF` and -DMUJOCO_ENABLE_AVX=OFF`) but I still had the same discrepancy so this did not solve my problem.

### Steps for reproduction

Here is an example of script giving 

1. Install mujoco on Avx-512 and non-Avx-512 computer.
2. Run the code below.
3. On Avx-512 computer I get `Sum of rewards is 991.5785945990693335261312313377857208251953125 and std is 0.17720032467718153` on non-Avx-512 I get `Sum of rewards is 988.127777684934017088380642235279083251953125 and std is 0.06265423998843093`

### Minimal model for reproduction

### Code required for reproduction

```python
import gymnasium as gym
import numpy as np
from stable_baselines3 import PPO, SAC
from stable_baselines3.common.env_util import make_vec_env
from stable_baselines3.common.utils import set_random_seed
import torch as th

th.use_deterministic_algorithms(True)
th.set_default_tensor_type(th.FloatTensor)

# Instantiate the environment, here a robotic Ant environment
env = gym.wrappers.DtypeObservation(gym.make('Ant-v5'), dtype=np.float64)

SEED = 3141592
# Set the random seed for stable-baselines3 and pytorch
set_random_seed(SEED)

# Uncomment what follows to retrain the model
model = PPO("MlpPolicy", env, verbose=1, seed = SEED)
model.learn(total_timesteps=5012)

# Check reproducibility of an initialization
obs, info = env.reset(seed=SEED)
print(f"Initial observation is {obs}")
print("")

env.action_space.seed(SEED)

df = pd.DataFrame()
obs, info = env.reset(seed=SEED)
env.action_space.seed(SEED)

# Evaluate
rews = []
for f in range(1000):
    action, _states = model.predict(obs,deterministic=True)
    obs, rewards, term, trunc, _ = env.step(action)
    rews.append(rewards)
    if term or trunc:
        break

print(f"Mean reward is {format(np.mean(rews), '.60g')} and std is {np.std(rews)}") 
```

### Confirmations

- [x] I searched the [latest documentation](https://mujoco.readthedocs.io/en/latest/overview.html) thoroughly before posting.
- [x] I searched previous [Issues](https://github.com/google-deepmind/mujoco/issues) and [Discussions](https://github.com/google-deepmind/mujoco/discussions), I am certain this has not been raised before.

yuvaltassa, 2025-05-29:
This is documented and working as intended. I suppose we should clarify in the docs that x86 with/out AVX are considered different architectures?

https://mujoco.readthedocs.io/en/latest/computation/index.html#reproducibility

closed 2025-05-29 by yuvaltassa

TimotheeMathieu, 2025-05-29:
Hmm, I am talking about avx-512 against avx-2 not avx against non-avx, but if you say this is intended then ok.

saran-t, 2025-05-29:
Just to make sure that I'm understanding this correctly: you're saying that if you `pip install mujoco` on a machine with AVX-512 vs a machine without AVX-512, you see different results?

Can you be more specific in terms of what processors are involved?

I don't know much about PyTorch but looking at the documentation page for `use_deterministic_algorithms` it says `given the same input, and when run on the same software and hardware, always produce the same output`, so it would seem that they don't guarantee reproducibility across different machines.