google-deepmind/mujoco #2674: Supporting PBR

DjKesu, 2025-06-11:
### The feature, motivation and pitch

Hi,

It would be great if Mujoco could start supporting different texture pbr maps, either through tags, or by directly supporting gltf/glb objects.

This would greatly open up a lot more types of simulations to happen inside of mujoco.

Thank you!

### Alternatives

Currrently, I'm using genesis and isaac sim as an alternative, but in terms of ease of use, and accuracy in simulations, I would love for this to exist in mujoco.

sriramsowmithri9807, 2025-06-15:
I’d like to work on this issue. Could @GJBoth please assign me?

saran-t, 2025-06-17:
@erez-tom to comment on the ORM texture type that was somewhat recently added. It's not plumbed into our built-in renderer at all, but we added it for our internal alternative renderer integration project. Perhaps that's already good enough for you?.

In the meantime, check out [this documentation page](https://mujoco.readthedocs.io/en/stable/XMLreference.html#material-layer).

siddancha, 2026-01-29:
Any updates/timeline on when filament might land? Thanks!