google-deepmind/mujoco #2697: [C API] Shared library build fails to link when using GCC.

davidhozic, 2025-06-20:
### Intro

Hi!

I am trying to compile MuJoCo as a shared library, however using the GCC toolchain, while it creates a shared library, that library fails to link afterwards.

### My setup

- Ubuntu 24.04.2 LTS
- MuJoCo main branch (3.3.4?)
- GCC:
  - Thread model: posix
  - Supported LTO compression algorithms: zlib zstd
  - gcc version 13.3.0 (Ubuntu 13.3.0-6ubuntu2~24.04) 

### What's happening? What did you expect?

When compiling the shared library with **GCC**, linking fails with undefined symbols:

```
test.c:(.text+0x1c): undefined reference to `mj_loadModel'
```
and also other functions when I try to use them.

Upon inspection with ``nm``, not all of the library file can be read, saying:

```nm: mujoco/build/lib/libmujoco.so: plugin needed to handle lto object
00000000000037d9 b completed.0
                 w __cxa_finalize
0000000000001540 t deregister_tm_clones
00000000000015b0 t __do_global_dtors_aux
0000000000002630 d __do_global_dtors_aux_fini_array_entry
00000000000037b0 d __dso_handle
0000000000002640 d _DYNAMIC
000000000000152c t _fini
00000000000015f0 t frame_dummy
0000000000002638 d __frame_dummy_init_array_entry
000000000000050c r __FRAME_END__
00000000000037b8 d _GLOBAL_OFFSET_TABLE_
                 w __gmon_start__
00000000000037d8 B __gnu_lto_slim
0000000000001510 t _init
                 w _ITM_deregisterTMCloneTable
                 w _ITM_registerTMCloneTable
0000000000001570 t register_tm_clones
00000000000037b0 d __TMC_END__
00000000000037b0 d __TMC_LIST__
```

Everything works with **CLANG**. 

### Steps for reproduction

CWD=$(pwd)
CMAKE_ARGS=(
    -DCMAKE_C_COMPILER:STRING=gcc
    -DCMAKE_CXX_COMPILER:STRING=g++
    -DBUILD_SHARED_LIBS:BOOL=ON
    -DMUJOCO_HARDEN:BOOL=OFF
    -DCMAKE_BUILD_TYPE:STRING=Release
    -DCMAKE_INTERPROCEDURAL_OPTIMIZATION:BOOL=ON
    -DMUJOCO_BUILD_EXAMPLES:BOOL=OFF
    -DCMAKE_EXE_LINKER_FLAGS:STRING=-Wl,--no-as-needed
)

cmake -B build -S . "${CMAKE_ARGS[@]}" -G "Unix Makefiles"
cmake --build build --parallel --target simulate --config=Release

### Minimal model for reproduction

Not relevant.

### Code required for reproduction

## Build MuJoCo:
```
CMAKE_ARGS=(
    -DCMAKE_C_COMPILER:STRING=gcc
    -DCMAKE_CXX_COMPILER:STRING=g++
    -DBUILD_SHARED_LIBS:BOOL=ON
    -DMUJOCO_HARDEN:BOOL=OFF
    -DCMAKE_BUILD_TYPE:STRING=Release
    -DCMAKE_INTERPROCEDURAL_OPTIMIZATION:BOOL=ON
    -DMUJOCO_BUILD_EXAMPLES:BOOL=OFF
    -DCMAKE_EXE_LINKER_FLAGS:STRING=-Wl,--no-as-needed
)

cmake -B build -S . "${CMAKE_ARGS[@]}"
cmake --build build --parallel --target mujoco --config=Release
```

## Compile the user application (test.c):
```
gcc test.c \
    -L mujoco_rust/mujoco/build/lib/ \
    -lmujoco \
    -I mujoco_rust/mujoco/include/mujoco/ \
    -I mujoco_rust/mujoco/include/ \
    -Wl,-rpath,$PWD/mujoco_rust/mujoco/build/lib \
    -o test
```

## test.c:
```c
#include "mujoco.h"

int main() {
    mjModel *model = mj_loadModel("dummy.xml", NULL);
    return 0;
}
```

### Confirmations

- [x] I searched the [latest documentation](https://mujoco.readthedocs.io/en/latest/overview.html) thoroughly before posting.
- [x] I searched previous [Issues](https://github.com/google-deepmind/mujoco/issues) and [Discussions](https://github.com/google-deepmind/mujoco/discussions), I am certain this has not been raised before.

traversaro, 2025-06-20:
Did you tried to compile the user application with CMake instead of using raw gcc?

davidhozic, 2025-06-20:
Same thing if I try to compile test.c via CMake (I replaced main.cc of simulate with my file):

``ld.lld: error: undefined symbol: main``

```c
#include "mujoco/mujoco.h"
#include "stdio.h"

int main() {
    mjModel *model = mj_loadModel("dummy.xml", NULL);
    printf("%p", (void *) model);  // prevent unused error
    return 0;
}
```

davidhozic, 2025-06-20:
> > Did you tried to compile the user application with CMake instead of using raw gcc?
> 
> I tried to compile the simulate program (viewer) with CMake, but then it fails with an undefined reference to main.

Same thing if I try to compile test.c via CMake (I replaced main.cc of simulate with my file):

``ld.lld: error: undefined symbol: main``

```c
#include "mujoco/mujoco.h"
#include "stdio.h"

int main() {
    mjModel *model = mj_loadModel("dummy.xml", NULL);
    printf("%p", (void *) model);  // prevent unused error
    return 0;
}
```

yuvaltassa, 2025-06-21:
Our GH CI builds with gcc, you should be able to see what we do there, as recommended in the docs

https://mujoco.readthedocs.io/en/latest/programming/index.html#building-from-source

closed 2025-06-21 by yuvaltassa

YHLin117, 2025-08-28:
> > > Did you tried to compile the user application with CMake instead of using raw gcc?
> > 
> > 
> > I tried to compile the simulate program (viewer) with CMake, but then it fails with an undefined reference to main.
> 
> Same thing if I try to compile test.c via CMake (I replaced main.cc of simulate with my file):
> 
> `ld.lld: error: undefined symbol: main`
> 
> #include "mujoco/mujoco.h"
> #include "stdio.h"
> 
> int main() {
>     mjModel *model = mj_loadModel("dummy.xml", NULL);
>     printf("%p", (void *) model);  // prevent unused error
>     return 0;
> }

hello,have you solved this problem?