google-deepmind/mujoco #2724: Move mjcb_.* callbacks into mjModel

maximilian-maisel-bl, 2025-07-08:
### The feature, motivation and pitch

Multiple different MuJoCo simulation models in one process are not cleanly separated
because the installed `mjcb_.*` callbacks from `mujoco.h` are global.

Please move these function pointers into the `mjModel` struct to provide separation
between different simulation contexts. All callbacks except `mjcb_time` already require a pointer to a `mjModel` struct so this change should not be too invasive.

### Alternatives

It is possible to swap the pointers before each call to `mj_step`.
However, this results in some kind of "Big Simulator Lock".

saran-t, 2025-08-06:
`mjcb_` things should be considered deprecated for all but the simplest of use cases. If you need separation by models then then your use case already extends beyond what I meant by simple.

Can you use [plugins](https://mujoco.readthedocs.io/en/stable/programming/extension.html#engine-plugins)?

maximilian-maisel-bl, 2025-08-07:
At the moment, I'm calculating all actuator torques before calling `mj_step`
in a loop in the simulation thread. For our current use-case this seems to work fine.
Plugins are looking promising if this approach should be no longer sufficient in future.

I think the documentation should be improved and clearly mention that these callbacks are
deprecated and plugins are the preferred way.

Currently, the [documentation](https://mujoco.readthedocs.io/en/stable/programming/simulation.html), chapter `Simulation loop`
states that "The default (and recommended) way to control the system is to implement a
control callback, for example [...] To install this callback, we simply assign it to the global
control callback pointer mjcb\_control:". This documentation does not mention anything about
"deprecation" or "plugins".