google-deepmind/mujoco #2741: Bug in differentiable physics tutorial notebook

MartinOpat, 2025-07-10:
### Intro

Hi!

I am a (graduate student) at the University of Groningen, I want to use MuJoCo for my research on locomotion via differentiable simulation. However, when trying to run the **differentiable physics** tutorial [notebook](https://colab.research.google.com/github/google-deepmind/mujoco/blob/main/mjx/training_apg.ipynb) linked in the official README.md

### My setup

I simply opened the tutorial jupyter [notebook](https://colab.research.google.com/github/google-deepmind/mujoco/blob/main/mjx/training_apg.ipynb) in google colab, set the "runtime type" to "T4 GPU" and pressed "run all". 

### What's happening? What did you expect?

I get the following error in the *Imitation Learning via FoPG's* cell:

```
/usr/local/lib/python3.11/dist-packages/brax/io/mjcf.py:480: UserWarning: Brax System, piplines and environments are not actively being maintained. Please see MJX for a well maintained JAX-based physics engine: https://github.com/google-deepmind/mujoco/tree/main/mjx. For a host of environments that use MJX, see: https://github.com/google-deepmind/mujoco_playground.
  warnings.warn(

---------------------------------------------------------------------------

AttributeError                            Traceback (most recent call last)

[/usr/local/lib/python3.11/dist-packages/mujoco/mjx/_src/types.py](https://localhost:8080/#) in __getattr__(self, name)
   1135       impl_instsance = object.__getattribute__(self, '_impl')
-> 1136       val = getattr(impl_instsance, name)
   1137       warnings.warn(

AttributeError: 'DataJAX' object has no attribute 'pos'

During handling of the above exception, another exception occurred:

AttributeError                            Traceback (most recent call last)

15 frames

    [... skipping hidden 13 frame]

    [... skipping hidden 23 frame]

    [... skipping hidden 10 frame]

    [... skipping hidden 7 frame]

[/usr/local/lib/python3.11/dist-packages/mujoco/mjx/_src/types.py](https://localhost:8080/#) in __getattr__(self, name)
   1143     except AttributeError:
   1144       # raise the standard exception
-> 1145       raise AttributeError(  # pylint: disable=raise-missing-from
   1146           f"'{type(self).__name__}' object has no attribute '{name}'"
   1147       )

AttributeError: 'State' object has no attribute 'pos'
```

If I manually specify python package versions from the time of the last edit of the notebook, i.e.:
```
!pip install mujoco==3.2.4
!pip install mujoco_mjx==3.2.4
!pip install brax==0.11.0
```
then the error disappears. However, on the recent versions of mujoco (3.3.3 or 3.3.4) the notebook does not run.

### Steps for reproduction

1. Open [notebook](https://colab.research.google.com/github/google-deepmind/mujoco/blob/main/mjx/training_apg.ipynb) 
2. Select a gpu runtime
3.  I had to change `jax.lax.Precision.HIGH` to `"high"` in the line `config.update('jax_default_matmul_precision', jax.lax.Precision.HIGH)` in cell *Import MuJoCo, MJX, and Brax* to avoid a `ValueError: new enum value must be None or in ['default', 'high' ...` error.
4. "Run all"

### Minimal model for reproduction

N/A

### Code required for reproduction

No changes other than step 3. in "Steps for reproduction" were done to the notebook.

### Confirmations

- [x] I searched the [latest documentation](https://mujoco.readthedocs.io/en/latest/overview.html) thoroughly before posting.
- [x] I searched previous [Issues](https://github.com/google-deepmind/mujoco/issues) and [Discussions](https://github.com/google-deepmind/mujoco/discussions), I am certain this has not been raised before.

btaba, 2025-07-10:
@Andrew-Luo1

Andrew-Luo1, 2025-07-14:
Fixed by https://github.com/google-deepmind/mujoco/pull/2743 good luck with your research :)