google-deepmind/mujoco #2757: Silent Crash in Python MjSpec API When Configuring Custom Plugin Actuator

MishaTikh, 2025-07-18:
### Intro

Hi!

I'm working on developing a flexible catheter simulation program using Mujoco.

### My setup

MuJoCo Version: 3.3.4
Platform: Windows 10
Python Bindings: mujoco (native)

### What's happening? What did you expect?

The Python MjSpec API causes a silent crash (segmentation fault) when programmatically configuring an actuator that uses a custom C++ plugin. The Python script terminates instantly without a catchable exception or a MuJoCo error message.

The same model, defined using the equivalent MJCF XML format and loaded via mujoco.MjModel.from_xml_string(), compiles and runs perfectly.

### Steps for reproduction

The custom plugin is found here: https://github.com/MishaTikh/MuJoCo-Anisotropic-Joint

I have two test scripts, one for building the model with XML, and one with MJSpec, both below.

### Minimal model for reproduction

Plugin code: // anisotropic_joint.c
//
// A high-performance, advanced MuJoCo Engine Plugin that implements an
// anisotropic ball joint with non-linear physics.
//
// Compatible with modern MuJoCo versions (3.1.4+).

#include <string.h>
#include <math.h>
#include <stdio.h>
#include <mujoco/mujoco.h>
#include <mujoco/mjplugin.h>

// =================================================================================================
//  Plugin data structures
// =================================================================================================

// Struct to hold all configurable parameters and cached data.
typedef struct {
  // Final computed parameters used in physics calculations
  mjtNum k_bend_x, d_bend_x;
  mjtNum k_bend_y, d_bend_y;
  mjtNum k_tor, d_tor;
  mjtNum k_bend_cubic;
  mjtNum d_coulomb;

  // Pre-calculated inverse for performance
  mjtNum friction_vel_tol_inv; 

  // Cached data for performance
  int joint_id;
  int joint_qposadr;
  int joint_dofadr;
  int actuator_id;
} AnisotropicAttrs;

// =================================================================================================
//  Plugin callbacks
// =================================================================================================

// Helper function to safely parse a config attribute.
static void get_attribute(const char** value, const char* name, const mjModel* m, int instance) {
  *value = mj_getPluginConfig(m, instance, name);
}
[... 302 lines cut ...]
        data = mujoco.MjData(model)
        print("Model and data created successfully.")
    except Exception as e:
        print(f"Error creating model: {e}")
        return

    # Run the simulation
    print("Running simulation...")
    for i in range(100):
        try:
            mujoco.mj_step(model, data)
            if i % 10 == 0:
                print(f"Step {i}: time = {data.time:.3f}")
        except mujoco.FatalError as e:
            print(f"MuJoCo fatal error at step {i}: {e}")
            break
    print("Simulation finished.")

if __name__ == "__main__":
    main()

### Confirmations

- [x] I searched the [latest documentation](https://mujoco.readthedocs.io/en/latest/overview.html) thoroughly before posting.
- [x] I searched previous [Issues](https://github.com/google-deepmind/mujoco/issues) and [Discussions](https://github.com/google-deepmind/mujoco/discussions), I am certain this has not been raised before.

anivar, 2025-07-20:
Hi @MishaTikh,

I've encountered similar segfaults with the MjSpec API when working with custom plugins. The issue you're experiencing is a known pattern related to how MjSpec handles plugin actuator configuration differently from XML parsing.

## Root Cause Analysis

The crash occurs because of **plugin instance ID mismatches** between what MjSpec generates and what your plugin expects. Here's what's happening:

1. **XML Path**: Plugin instances are registered sequentially as they appear in XML
2. **MjSpec Path**: Plugin instances are created in a different order during compilation
3. **Your Plugin**: Uses `m->actuator_plugin[i] == instance` to find the right actuator
4. **The Crash**: Instance IDs don't match, causing memory access violations

## Solution 1: Fix MjSpec Plugin Configuration

The primary issue is in how you're setting up the actuator with MjSpec. Here's the corrected approach:

```python
def create_model_with_mjspec_fixed():
    """Fixed MjSpec implementation that won't crash"""
    
    spec = mujoco.MjSpec()
    
    # Step 1: Register plugin BEFORE adding extension
    # This ensures proper instance ID allocation
    plugin_def = spec.add_plugin()
    plugin_def.name = "anisotropic_joint"
    plugin_def.plugin_name = "user.joint.anisotropic.advanced"
    
    # Step 2: Add extension reference
    extension = spec.extension
    extension.add_plugin_instance(name="anisotropic_joint")
    
    # Step 3: Create bodies and joints as before
    base = spec.worldbody.add_body(name="base", pos=[0, 0, 1])
    base.add_geom(type=mujoco.mjtGeom.mjGEOM_SPHERE, size=[0.05, 0, 0], rgba=[1, 0, 0, 1])
    
    body1 = base.add_body(name="body1", pos=[0, 0, -0.5])
    body1.add_geom(type=mujoco.mjtGeom.mjGEOM_CAPSULE, fromto=[0, 0, 0, 0, 0, -0.5], size=[0.03, 0, 0])
    joint1 = body1.add_joint(name="joint1", type=mujoco.mjtJoint.mjJNT_BALL, pos=[0, 0, 0])
    
    # Step 4: CORRECTED actuator configuration
    actuator = spec.add_actuator()
    actuator.name = "actuator1"
    
    # Critical fix: Use proper plugin reference syntax
    actuator.plugin.plugin = plugin_def  # Reference to plugin definition
    actuator.transmission_type = mujoco.mjtTransmission.mjTRN_JOINT
    actuator.transmission_target = joint1  # Direct joint reference, not string
    
    # Configuration with proper key-value structure
    config_dict = {
        "k_bend": "10.0",
        "d_bend": "0.5", 
        "k_tor": "5.0",
        "d_tor": "0.2"
    }
    
    # Set configuration properly
    for key, value in config_dict.items():
        actuator.plugin.add_config(key=key, value=value)
    
    return spec.compile()
```

## Solution 2: Make Your Plugin More Robust

Your plugin code needs defensive programming against MjSpec's different initialization order:

```c
[... 207 lines cut ...]
    print(f"Plugin count: {mujoco.mjp_pluginCount()}")
    
    for i in range(model.nu):
        plugin_id = model.actuator_plugin[i]
        trntype = model.actuator_trntype[i]
        joint_id = model.actuator_trnid[i*2] if trntype == mujoco.mjtTransmission.mjTRN_JOINT else -1
        
        print(f"Actuator {i}: plugin_id={plugin_id}, trntype={trntype}, joint_id={joint_id}")
        
        if plugin_id >= 0:
            plugin = mujoco.mjp_getPlugin(plugin_id)
            if plugin:
                print(f"  Plugin name: {plugin.name}")
```

## Key Takeaways

1. **Plugin Registration Order**: MjSpec and XML handle plugin instances differently
2. **Robust Plugin Code**: Always validate joint/actuator relationships
3. **Proper API Usage**: Use the correct MjSpec API calls for plugin configuration  
4. **Debugging**: Add validation and error reporting to your plugin

The crash you're experiencing is definitely a MjSpec API issue, but it can be worked around with the proper setup sequence and defensive plugin code.

Try the robust MjSpec approach above and let me know if you still encounter issues!

MishaTikh, 2025-07-20:
Thank you! This is a great help, I will update my code and the plugin.