google-deepmind/mujoco #2761: Support body transmission type in actuator/plugin.

frankpzh, 2025-07-22:
### The feature, motivation and pitch

I'm writing an actuator plugin for vacuum grippers, which creates suction forces to objects nearby in a fixed direction.
The plugin instance needs to be attached to a body rather than a joint, like what actuator/adhesion does.

I see actuator/general and actuator/adhesion has body transmission, but not actuator/plugin. Looks to me it's natual to let actuator/plugin have it as well.

I'm able to implement the plugin for vacuum grippers, besides a few changes to the mujoco code base:
https://github.com/google-deepmind/mujoco/pull/2760

### Alternatives

Giving one plugin instance for each vacuum gripper, and use instance config to specify the body name of each vacuum head.
I haven't tried that. It would be hard because the contact moment matrix is created before the plugin is called (in mj_transmission), and update the sparse matrix could be a mess.

yuvaltassa, 2025-08-11:
I see you opened and closed a PR?

Seems fairly straightforward?

frankpzh, 2025-10-01:
@yuvaltassa It takes me a while to figure out the CLA stuff.
I think the PR is ready for review now.

Thanks a lot.