google-deepmind/mujoco #2775: OpenUSD materials are not being exported correctly

oursland, 2025-07-28:
### Intro

I am experimenting with the new C++ exporter for OpenUSD files (`src/experimental/usd/`) and the exported `.usda` files do not have valid materials.  Instead of the colors and textures in the MuJoCo XML, it is just matte gray or fully reflective shiny surfaces.

### My setup

1. I have compiled MuJoCo `main` (https://github.com/google-deepmind/mujoco/commit/e4eb6545737e4812656c4e3e4c9b177f3678026e) for macOS.
2. I have set `PXR_PLUGINPATH_NAME` to the location of the plugins
3. In `model/car/`, I run `usdcat car.xml > car.usda`

### What's happening? What did you expect?

When the `car.usda` file is imported into Blender a scene similar to `simulate`'s is expected:

[image]

Actual display is:

[image]

### Steps for reproduction

1. Compiled MuJoCo `main` (https://github.com/google-deepmind/mujoco/commit/e4eb6545737e4812656c4e3e4c9b177f3678026e) for macOS with `-DMUJOCO_WITH_USD=ON`
2. Set `PXR_PLUGINPATH_NAME` to the location of the plugins
3. In `model/car/`, I run `usdcat car.xml > car.usda`
4. Import the `car.usda` file in Blender

### Confirmations

- [x] I searched the [latest documentation](https://mujoco.readthedocs.io/en/latest/overview.html) thoroughly before posting.
- [x] I searched previous [Issues](https://github.com/google-deepmind/mujoco/issues) and [Discussions](https://github.com/google-deepmind/mujoco/discussions), I am certain this has not been raised before.

havess, 2025-07-29:
Hey! Thanks for checking this out, looks a bug with support of geom rgba attribute. I'll look into it tomorrow.

In the meantime can you post the `car.usda` here just to double check nothing else is happening?

oursland, 2025-07-29:
Sure, here it is!

[car.usda.txt](https://github.com/user-attachments/files/21496880/car.usda.txt)

havess, 2025-07-30:
There was a bug where we weren't authoring the `primvars:displayColor` for site geometry, which I have a fix for that should be in soon. Otherwise this looks like a Blender bug where they aren't parsing the primvar correctly for geometry primitives.

Here are screenshots from HdStorm in usdview and Karma in Houdini where the colors are correct.

I'll keep this up until the site displayColor fix has landed.

[image]
[image]

havess, 2025-07-31:
I've commited the fix for Site displayColor primvar authoring in the MJCF SdfFileFormat plugin.

Let me know if you run into any other issues/missing features with USD support in MuJoCo.

oursland, 2025-07-31:
Unfortunately, I cannot confirm the fix.  Attached is the new `car.usda` file, which is substantially larger than the last, but it doesn't render as expected in either Blender or `usdview`.

[car.usda.txt](https://github.com/user-attachments/files/21537927/car.usda.txt)

havess, 2025-08-01:
Those green crosses are Sites in MuJoCo, and when we convert to USD we set the render purpose to “guide”. There should be a toggle under the Display menu dropdown that lets you enable rendering of “guide” and you’ll see the crosses.

closed 2025-08-05 by kbayes