google-deepmind/mujoco #2785: Scene Compilation Error in MuJoCo 3.3.4/3.3.3 (Works in 3.3.2)

HoangGiang93, 2025-08-03:
### Intro

Hi!

I am a researcher at AICOR Institute for Artificial Intelligence, University of Bremen, I use MuJoCo for my [research](https://github.com/Multiverse-Framework/Multiverse).

### My setup

MuJoCo>=3.3.2, Python3.12, Ubuntu>=20.04

### What's happening? What did you expect?

I've discovered a bug in MuJoCo versions 3.3.4 and 3.3.3 that doesn't appear in version 3.3.2.

The issue occurs in a scene containing laser sensors, which I implemented in two variants: one using replicate on sites, and one without. When compiling the scene with spec and exporting it to a file, both 3.3.4 and 3.3.3 produce incorrect results.

In the full version, bodies are excluded and only the sites are exported. In the replicate version, the body containing the sites isn't parsed at all. I’ve attached the models to reproduce the issue, along with a video to demonstrate the bug more clearly.

https://github.com/user-attachments/assets/651f9594-3db7-445a-ab9c-57a653fcf579

### Steps for reproduction

1. Load the model below.

[bug.zip](https://github.com/user-attachments/files/21567675/bug.zip)

2. Run the code below.

```
import mujoco

print(mujoco.mj_version())

spec = mujoco.MjSpec.from_file('./scene_rep.xml')
spec.compile()
spec.to_file('./scene_rep_compiled.xml')

spec = mujoco.MjSpec.from_file('./scene_full.xml')
spec.compile()
spec.to_file('./scene_full_compiled.xml')
```

### Minimal model for reproduction

<mujoco model="scene_rep">
  <compiler angle="radian" autolimits="true" fusestatic="true"/>

  <option integrator="implicitfast"/>

  <worldbody>
    <body name="base_link" pos="2.0 2.0 0">
      <body name="laser_mount" pos="0.202 0 0.4" euler="0 0 0">
        <body name="laser_body" euler="1.5707 0 0">
          <replicate count="721" euler="0 0.004363 0" sep="_">
            <site name="laser_site" />
          </replicate>
        </body>
      </body>
    </body>
  </worldbody>

  <sensor>
    <rangefinder name="laser_0" site="laser_site_000" />
    <rangefinder name="laser_1" site="laser_site_001" />
    <rangefinder name="laser_2" site="laser_site_002" />
    <rangefinder name="laser_3" site="laser_site_003" />
    <rangefinder name="laser_4" site="laser_site_004" />
    <rangefinder name="laser_5" site="laser_site_005" />
    <rangefinder name="laser_6" site="laser_site_006" />
    <rangefinder name="laser_7" site="laser_site_007" />
[... 720 lines cut ...]
      
      <geom name="arrow_shaft" type="cylinder" size="0.05 0.3" rgba="1 0 0 1" pos="0.1 0 0.01" euler="0 1.57 0" mass="0" contype="0" conaffinity="0"/>
    </body>
  </worldbody>

</mujoco>

### Code required for reproduction

import mujoco

print(mujoco.mj_version())

spec = mujoco.MjSpec.from_file('./scene_rep.xml')
spec.compile()
spec.to_file('./scene_rep_compiled.xml')

spec = mujoco.MjSpec.from_file('./scene_full.xml')
spec.compile()
spec.to_file('./scene_full_compiled.xml')

### Confirmations

- [x] I searched the [latest documentation](https://mujoco.readthedocs.io/en/latest/overview.html) thoroughly before posting.
- [x] I searched previous [Issues](https://github.com/google-deepmind/mujoco/issues) and [Discussions](https://github.com/google-deepmind/mujoco/discussions), I am certain this has not been raised before.

HoangGiang93, 2025-08-06:
> Hi, I'm not sure why it was working in 3.3.2 but replicate should do the sensor replication for you. Have you tried
> 
> ```
> <mujoco model="scene_rep">
>   <compiler angle="radian" autolimits="true" fusestatic="true"/>
> 
>   <option integrator="implicitfast"/>
> 
>   <worldbody>
>     <body name="base_link" pos="2.0 2.0 0">
>       <body name="laser_mount" pos="0.202 0 0.4" euler="0 0 0">
>         <body name="laser_body" euler="1.5707 0 0">
>           <replicate count="721" euler="0 0.004363 0" sep="_">
>             <site name="laser_site" />
>           </replicate>
>         </body>
>       </body>
>     </body>
>   </worldbody>
> 
>   <sensor>
>     <rangefinder name="laser" site="laser_site"/>
>   </sensor>
> 
>   <worldbody>
>     <body name="cylinder_with_arrow" pos="5 2 0" mocap="true">  
>       <inertial pos="0 0 0" mass="1e-9" diaginertia="1e-9 1e-9 1e-9"/>
>       
>       <geom type="cylinder" size="0.3 0.8" rgba="0 0 1 1" pos="0 0 0.8" mass="0" contype="0" conaffinity="0"/>
>       
>       <geom name="arrow_shaft" type="cylinder" size="0.05 0.3" rgba="1 0 0 1" pos="0.1 0 0.01" euler="0 1.57 0" mass="0" contype="0" conaffinity="0"/>
>     </body>
>   </worldbody>
> 
> </mujoco>
> ```

Thank your for your comment, I just learned something new :). However, it conducted same result as my scene_rep.xml.

quagla, 2025-08-11:
Can you confirm you still have the issue with 3.3.5? I have just tried it and it works okay for me.