google-deepmind/mujoco #2801: Using SDF plugin in mujoco 3.3.5

lanfeiying, 2025-08-12:
### Intro

Hi!

I am a (researcher) at University, I use MuJoCo for my research on dynamic simulation.
The newest version mujoco 3.3.5 was release, and the sdf plugin is supported natively. I have some issue in using it.

### My setup

Here is the minimal example of MJCF file:
```xml
<mujoco model="iiwa14">
  <compiler angle="radian" meshdir="assets" autolimits="true"/>

  <option integrator="implicitfast" iterations="200" sdf_iterations="40" sdf_initpoints="80" ccd_iterations="100"/>

  <extension>
    <plugin plugin="mujoco.sdf.sdflib">
        <instance name="sdf">
            <config key="aabb" value="0"/>
        </instance>
    </plugin>
  </extension>

  <asset>
    <mesh name="pin_20" file="pin_20.STL"><plugin instance="sdf"></plugin></mesh>
    <mesh name="hole_22" file="hole_22.STL"><plugin instance="sdf"/></mesh>
  </asset>

  <worldbody>
    <light name="top" pos="0 0 2" mode="trackcom"/>

    <body name="hole" pos="0.5 0 0">
      <inertial mass="2" pos="0 0 0" diaginertia="0.05 0.05 0.03"/>
      <geom name="hole_22" type="sdf" mesh="hole_22" contype="1" conaffinity="1"></geom>
    </body>

    <body name="base" childclass="iiwa">
        <body name="link_tool" pos="0.0 0.0 0.033">
            <geom name="tool" type="sdf" mesh="pin_20" contype="1" conaffinity="1"></geom>       
            <inertial mass="1.0" pos="0 0 0.033" diaginertia="0.001 0.001 0.0003" />        
        </body>
    </body>
  </worldbody>

</mujoco>

```

### What's happening? What did you expect?

I am using C++ writing code using Mujoco 3.3.5. When I run this, and when the pin is approaching to the hole, it shows an error: to many contact points. I put this xml file in `mujoco/bin/simulate` application, and press `h` key to visualise the convex hull, the hole model is still filled by convex hull.

I searched the documentation, but the introduction for using the general sdf plugin is scarce. May I ask what is the example code for using the general sdf plugin in MJCF? What did I miss?

### Steps for reproduction

- I run the c++ programming using Mujoco library.
- I also tested the GUI programme provided by the package `mujoco/bin/simulate`.

### Confirmations

- [x] I searched the [latest documentation](https://mujoco.readthedocs.io/en/latest/overview.html) thoroughly before posting.
- [x] I searched previous [Issues](https://github.com/google-deepmind/mujoco/issues) and [Discussions](https://github.com/google-deepmind/mujoco/discussions), I am certain this has not been raised before.

lanfeiying, 2025-08-12:
> Yes it should be fine, but I would update your XML first by removing the plugin and re-test it.

I updated my XML file by removing all sdf plugin codes. It works now.

Would you recommend using SDF plugin for mesh-sdf collision simulation? Is there any solution to the issue of "too many contact points" for mesh-sdf collision ?

quagla, 2025-08-12:
The plugin should only be used if you have your own analytic SDF.

For mesh-SDF collisions I have a WIP solution which is under review internally and will be slightly more restrictive in the sense that it will use the convex hull of the mesh (as it's the case for all mesh collisions in mujoco).

lanfeiying, 2025-08-12:
In my simulation, I have concave STL file which cannot be converted into analytic SDF. So I am using the general SDF plugin which uses the volumentric representation.

Shall I continue to use this in the future? Will the general SDF cause issues in collision?

lanfeiying, 2025-08-12:
> The plugin should only be used if you have your own analytic SDF.
> 
> For mesh-SDF collisions I have a WIP solution which is under review internally and will be slightly more restrictive in the sense that it will use the convex hull of the mesh (as it's the case for all mesh collisions in mujoco).

I am using SDF-SDF collision, when the two part contacts with two planes, the simulator still prompts error: too many contact points. 

I am wondering if there are potential solutions to this issue? Even for flat plane contact for SDF-SDF collision?