google-deepmind/mujoco #2806: Named element accessors for `mjx.Model`

hartikainen, 2025-08-16:
### The feature, motivation and pitch

Both `mujoco.MjModel` and `mujoco.MjSpec` have a nice API where their elements can be accessed by name with methods like `mujoco.MjModel.{body,geom,joint,actuator}`. I notice myself constantly missing similar API for `mjx.Model`. For example, most recently, I tried using the playground environment interface together with domain randomization. Unfortunately, [`BraxDomainRandomizationVmapWrapper`](https://github.com/google-deepmind/mujoco_playground/blob/25f570d83fbf13519ca854f08c245b43ebab10bc/mujoco_playground/_src/wrapper.py#L166) only passes `self.mjx_model` into the `randomization_fn`, which makes it hard to access its elements with names.

### Additional context

Here's a simple test case I wish would pass:
<details><summary>Details</summary>

```python
"""Test documentation."""

import textwrap

from absl.testing import absltest
from absl.testing import flagsaver
from absl.testing import parameterized
import mujoco
from mujoco import mjx

class UnitTests(parameterized.TestCase):
    def test_add(self) -> None:
        model = mujoco.MjModel.from_xml_string(textwrap.dedent("""
            <mujoco model="test">
              <worldbody>
                <geom name="plane" type="plane" size="1 1 1"/>
                <body name="body" pos="0 0 0">
                  <joint name="joint" type="slide" axis="1 0 0" range="-5 5"/>
                  <geom name="box" type="box" size=".2 .1 .1" rgba=".9 .3 .3 1"/>
                </body>
              </worldbody>
              <actuator>
                <motor joint="joint" name="motor"/>
              </actuator>
            </mujoco>
        """))
        modelx = mjx.put_model(model)

        self.assertEqual(model.body("body").id, modelx.body("body").id)
        self.assertEqual(model.joint("joint").id, modelx.joint("joint").id)
        self.assertEqual(model.actuator("motor").id, modelx.actuator("motor").id)
        self.assertEqual(model.geom("plane").id, modelx.geom("plane").id)
        self.assertEqual(model.geom("box").id, modelx.geom("box").id)

if __name__ == "__main__":
    absltest.main()
```

</details>

Yuvraj-cyborg, 2025-12-23:
Hey @hartikainen  I would like to take on this issue  !

SebasNorenaR, 2026-11-09:
Closing as `mujoco_playground` has since migrated to `reaf-sim` (`mujoco_warp`) where domain randomization has direct model/spec access.

closed 2026-11-09 by SebasNorenaR