google-deepmind/mujoco #2823: Inconsistent `qpos0` handling in the simulate viewer

MasonMcGill, 2025-08-27:
### Intro

Hi,

I'm a computational neuroscientist at HHMI Janelia. I use MuJoCo to simulate fruit fly biomechanics.

### My setup

MuJoCo 3.3.5 with Python 3.12 on Ubuntu 25.04.

### What's happening? What did you expect?

I'm trying to store a model file with a programmatically generated initial pose, and it seems like the simulate viewer (`mujoco/bin/simulate`) isn't handling saved models' `qpos0` buffers correctly. When testing a model with one free joint and 102 hinge joints, the free joint pose is correctly restored from `qpos0`, but the hinge joint poses are not.

### Steps for reproduction

1. Install Git and [uv](https://github.com/astral-sh/uv) if they aren't installed already.
2. Save the code snippet below as `demo-script.py`.
3. Run `uv run demo-script.py` to generate `model-with-keyframe.mjb` and `model-with-qpos0.mjb`.
4. Open `model-with-keyframe.mjb` in the simulate viewer included with MuJoCo 3.3.5.
5. Click "Pause" and then click the "0" next to "Key". This will show the pose returned by `lifted_leg_pose` in the script.
6. Open `model-with-qpos0.mjb` in the simulate viewer.
7. Click "Pause" and then click "reset". The free joint will be positioned correctly, but the hinge joints will be positioned incorrectly. And subsequently clicking the "0" next to "Key" will move the hinge joints to what looks like a sign-flipped version of their positions in the lifted leg pose.

### Minimal model for reproduction

### Code required for reproduction

```python
# /// script
# requires-python = "==3.12.*"
# dependencies = [
#   "absl-py==2.3.1",
#   "etils==1.13.0",
#   "fsspec==2025.7.0",
#   "glfw==2.9.0",
#   "importlib-resources==6.5.2",
#   "mujoco==3.3.5",
#   "numpy==2.3.2",
#   "pyopengl==3.1.10",
#   "typing-extensions==4.15.0",
#   "zipp==3.23.0",
# ]
# ///

from pathlib import Path
from shutil import move
from subprocess import run
from tempfile import TemporaryDirectory

import mujoco as mj

def main() -> None:
    download_model_spec_if_missing()
    write_model_with_keyframe()
    write_model_with_qpos0()

def download_model_spec_if_missing() -> None:
    output_path = Path("flybody")
    repo_url = "https://github.com/TuragaLab/flybody.git"
    commit_hash = "e1a6135c310c39291f4fb68d682f2fd0b05e0555"
    subdir = "flybody/fruitfly/assets"

    if not output_path.exists():
        with TemporaryDirectory() as repo:
            run(["git", "clone", repo_url, repo])
            run(["git", "-C", repo, "checkout", commit_hash, "--quiet"])
            move(Path(repo) / subdir, output_path)

def write_model_with_keyframe() -> None:
    model_spec = mj.MjSpec.from_file("flybody/floor.xml")
    model_spec.add_key(qpos=lifted_leg_pose())
    model = model_spec.compile()
    mj.mj_saveModel(model, "model-with-keyframe.mjb")

def write_model_with_qpos0() -> None:
    model = mj.MjModel.from_xml_path("flybody/floor.xml")
    model.qpos0[:] = lifted_leg_pose()
    mj.mj_setConst(model, mj.MjData(model))
    mj.mj_saveModel(model, "model-with-qpos0.mjb")

def lifted_leg_pose() -> list[float]:
    pose = 109 * [0.0]
    pose[:7] = [-0.0223, 0.0002, 0.0772, 0.9484, -0.2439, 0.1720, -0.1066]  # Free joint
    pose[98:104] = [0.0910, -0.4631, 0.3066, 0.6429, -1.1882, -0.2206]  # Leg joints
    return pose

if __name__ == "__main__":
    main()
```

### Confirmations

- [x] I searched the [latest documentation](https://mujoco.readthedocs.io/en/latest/overview.html) thoroughly before posting.
- [x] I searched previous [Issues](https://github.com/google-deepmind/mujoco/issues) and [Discussions](https://github.com/google-deepmind/mujoco/discussions), I am certain this has not been raised before.

yuvaltassa, 2025-09-01:
You can't just write into qpos0, many things depend on it.
Never modify qpos0 (other than by editing the XML or spec), only add keyframes.

I'm closing this but feel free to ask questions.

MasonMcGill, 2025-09-01:
Ok; thanks for letting me know.

Out of curiosity, what does `mj_setConst` do then? Based on the documentation my impression was that it propagates changes made to `qpos0` to other fields of a model, but if `qpos0` isn't meant to be user-writable I'm guessing that's not right.

MasonMcGill, 2025-09-02:
Ok that makes sense. Thanks for the explanation!

I think keyframes should work for what I'm trying to do (sharing 3D stills with collaborators).