google-deepmind/mujoco #2830: simulate.exe is not working on Windows !!

kshfire, 2025-09-01:
### Intro

Hi!

Please let me know why recent windows mujoco's simulate is not working.
- Mujoco version 3.2.5 is OK. However, it does not run in later versions.
What's the problem of it ?

Thanks.

### My setup

Windows 11 :  MuJoCo version 3.2.5's simulate.exe is working. But, later versions are not working.

### What's happening? What did you expect?

I want to use the recent verion's simulate.exe on the Windows.

### Steps for reproduction

Download windows mujoco. extract it and then try to run the simulate.exe.

### Confirmations

- [x] I searched the [latest documentation](https://mujoco.readthedocs.io/en/latest/overview.html) thoroughly before posting.
- [x] I searched previous [Issues](https://github.com/google-deepmind/mujoco/issues) and [Discussions](https://github.com/google-deepmind/mujoco/discussions), I am certain this has not been raised before.

kshfire, 2025-09-01:
Hi @yuvaltassa ,

Strangely, it works on my home PC, but not on my company PC.
- Company PC : When run it, only "MuJoCo version 3.3.5" is printed and terminated.
- It seems like the termination condition has been triggered, but I don't know why.  I'll check about it more, but if you have any hints, please let me know.
** Interestingly, MuJoCo version 3.2.5 is working...(???)

Anyway, thank you for your interest.

Thanks.

Vaidik1199, 2025-09-01:
Chat Gpt Said : 

It looks like the problem is not with MuJoCo itself but with the runtime environment on the company PC. The fact that simulate.exe works on your home machine and that older MuJoCo (3.2.5) works on the company machine suggests a missing dependency or configuration difference. A few things to check:

Graphics driver – Make sure the GPU drivers are up to date. Recent MuJoCo builds depend on modern OpenGL features, and if the driver is outdated simulate.exe may fail immediately after printing the version.

Microsoft Visual C++ redistributables – Install the latest Visual C++ runtime (x64). Newer builds may require libraries not needed in 3.2.5.

OpenGL / GPU availability – If you run on a system without proper GPU support, the executable can terminate right after start. Try forcing software OpenGL by setting the environment variable

set MJLIB=egl

before launching, just to test if it is a graphics issue.

Antivirus or security policies – On company PCs, antivirus or endpoint protection can silently block executables from running fully. Try running as administrator or checking if simulate.exe is being blocked.

Check for missing DLLs – You can use a tool like Dependency Walker on simulate.exe to see if any required DLLs are missing on that machine.

If none of these help, try running simulate.exe from a terminal instead of double-clicking so you can capture any error messages that appear right before termination.

kshfire, 2025-09-02:
Thanks @Vaidik1199,

My solution was : update visual studio !!!  Then, it worked. :)

Thanks.

closed 2025-09-02 by kshfire